tomvandeneede/p2pp #2: codec can't decode

mylife4aiurr, 2019-01-07:
```
Traceback (most recent call last):
  File "D:\Downloads\p2pp-master\p2pp.py", line 74, in <module>
    main(vars(arguments.parse_args()))
  File "D:\Downloads\p2pp-master\p2pp.py", line 64, in main
    args['silent']
  File "D:\Downloads\p2pp-master\p2pp\mcf.py", line 391, in generate
    gcode_file = opf.readlines()
  File "D:\Program Files\Python37\lib\encodings\cp1252.py", line 23, in decode
    return codecs.charmap_decode(input,self.errors,decoding_table)[0]
UnicodeDecodeError: 'charmap' codec can't decode byte 0x8d in position 5525: character maps to <undefined>
Press any key to continue . . .

```
the output file succeed:
https://drive.google.com/open?id=1Mx_sM-7jVnPhV8OMkWFbtEMSpDo6FMiv

mylife4aiurr, 2019-01-08:
The above error is related to python3 on window machines. I was able install and use script with python2.7 on windows

tomvandeneede, 2019-01-09:
Appropriate file encoding will be added in next release to both python 2 and 3 can be used

mylife4aiurr, 2019-01-09:
Looking forward to trying. Thanks for sharing by bridging the gap between Palette2's great hardware and slic3r's advanced slicing capabilities.

closed 2019-01-13 by tomvandeneede