tomvandeneede/p2pp #5: Side Wipe Travel Moves

totalitarian, 2019-01-16:
When transitioning from the print to start of the side wipe and from the end of the side wipe back to the print, no XY travel speed is defined, meaning it will crawl at whatever the last print speed was.

```
; Filament gcode
;P2PP Side Wipe
G1 X0.1 Y25
```

should be something like

```
; Filament gcode
;P2PP Side Wipe
G1 FXXX X0.1 Y25
```

where XXX = the speed setting in slic3r for no print moves (travel)

Same at the end of the purge

```
 PURGING FINISHED
G1 X150.398 Y132.959
G1 Z0.600
```

should be

```
 PURGING FINISHED
G1 FXXX X150.398 Y132.959
G1 Z0.600
```

totalitarian, 2019-01-16:
You'll also need to reset the speed back afterwards otherwise the print will be printer at non-print move speeds

totalitarian, 2019-01-16:
Also need to retract before and after the travel

totalitarian, 2019-01-16:
Also, why is this hard coded in?

`v.processedGCode.append("G1 X245 F200\n")`

tomvandeneede, 2019-01-21:
Did some changes in the dev version to address some of the issues.

tomvandeneede, 2019-02-12:
added possibility to add code before and after side wipe.

closed 2019-02-12 by tomvandeneede