tomvandeneede/p2pp #15: Shebang in P2PP.py not working under linux

deepthought-64, 2019-03-16:
I am running linux and pyhton 3 and when i execute P2PP.py (or for that matter p2pp.sh) i get the following error: 
`./p2pp.sh: ./P2PP.py: /usr/bin/pythonw: bad interpreter: No such file or directory`

For what I found after a quick search [1] the correct form should of the shebang in P2PP.py should be:
`#!/usr/bin/env python3`

Since I know very little about python I cannot say if that is the correct way but in my case it helped.

[[1]: Should I put #! (shebang) in Python scripts, and what form should it take?
](https://stackoverflow.com/questions/6908143/should-i-put-shebang-in-python-scripts-and-what-form-should-it-take)

tomvandeneede, 2019-04-14:
will setup a linux box and test this.   It might end up being a manual edit depending on the system...but give me some more time on this

tomvandeneede, 2019-09-13:
Ian,

sorry for not having looked into this yet... have been trying to keep things working with new version and new requirements thqt I hqve not mqnqged to get linux going

GarthSnyder, 2020-01-02:
I believe the OP is correct about this for Linux and macOS, but if the interpreter really is called **pythonw** (I'm assuming, for Python on Windows), **env python3** won't find that. There's probably some well-known stub you can insert to cover the common OSes.

tomvandeneede, 2020-01-02:
On this one need to further check.   Since there are only very few linux users this sort of went down the priority ladder… if anything there will be a third independent .sh script to cover this.  MACOS and Windows will need to remain independent

closed 2020-06-23 by tomvandeneede