tomvandeneede/p2pp #16: Cannot open "favicon.ico" under linux

deepthought-64, 2019-03-16:
When executing P2PP.py under linux I get the following stacktrace:

**Traceback (most recent call last):
  File "./P2PP.py", line 13, in <module>
    import p2pp.mcf as mcf
  File "/home/fabi/.anwendungen/p2pp-master/p2pp/mcf.py", line 13, in <module>
    import p2pp.gui as gui
  File "/home/fabi/.anwendungen/p2pp-master/p2pp/gui.py", line 26, in <module>
    root.iconbitmap(os.path.dirname(sys.argv[0])+'/favicon.ico')
  File "/usr/lib/python3.7/tkinter/__init__.py", line 1871, in wm_iconbitmap
    return self.tk.call('wm', 'iconbitmap', self._w, bitmap)
_tkinter.TclError: bitmap "./favicon.ico" not defined**

If I comment out the following line in `gui.py` it works:
`root.iconbitmap(os.path.dirname(sys.argv[0])+'\\favicon.ico')`

At first i thought it might have something to do with the backslash but when i replaced it with an normal slash it still wouldn't find the file.

I've added the following line into `gui.py` 
`print("Favicon file at: '"+os.path.dirname(sys.argv[0])+'/favicon.ico'+"'")`
to debug what the filename is. 
It prints `Favicon file at: './favicon.ico'` which looks as expected but I still have no clue why it is not finding the file.

tomvandeneede, 2019-04-14:
Will setup a linux box and test this

tomvandeneede, 2019-09-13:
for now icons will only be shown on windows systems

closed 2019-09-13 by tomvandeneede