tomvandeneede/p2pp #18: Wipe to waste object, Wipe to infill

Dragon2781, 2019-05-03:
Hi,
In your readme.md under 'Purpose' is listed that it is possible to wipe to a waste object or wipe to infill.
I can not find a description how to do that.
Can you please explain step by step or add this to the readme.md?
In my point of view this two points are THE reason to install P2PP instead of using Canvas or Chroma from Mosaic.

tomvandeneede, 2019-05-03:
Hi Dragon2781

it is beyond the scope of this project to support features that are taken from Slic3r PE.  
A simple search in google will answer most of these questions and there is a huge community that can help out on specific Slic3r PE questions.

Here is the first one I came across that may help you: https://www.youtube.com/watch?v=t11dzCPZqHc

However much I would like to be able to offer all this support in my readme, it would make it impossible to maintain it in the way I have it now on top of a day time job.

So I hope you accept that with this answer I close this issue.

Regards
Tom

closed 2019-05-03 by tomvandeneede

Dragon2781, 2019-05-03:
Hi, thank you for the video.
I'm new to Slic3r (coming from Cura). I did not know that this is a default setting there.
I thought this has to be done by the P2PP, sorry.
Thank you for your work :)