tomvandeneede/p2pp #19: SPLICEOFFSET only applies to first layer?

paul999, 2019-05-05:
Hi,

Thanks for your great work, I have been testing it for the last few days and it works really nicely 👍 

In the readme you mention the SPLICEOFFSET, where it says it applies to the first splace only (As far as I understand), is this correct? From canvas I remember that it applies too all splices, not just the first one, and to my understanding that would be the most logical as well, as otherwise there is no way to control where the purge is done on other splices as the first one.

(Not sure if the issues here is the best location to ask, please let me know if I should ask somewhere else instead)

tomvandeneede, 2019-05-05:
Well it has an effect on the first splice length only, but the offset continues throughout the entire print.

So the first length of filament will get an additional few mm as specified by the SplicaOffset so the filament color switch point is pushed a little further in the purge area.   This is to prevent an early color swap if the printer would be consuming more filament than expected.   This is similar to the % purge position in canvas.

Hope this answers your question.

You can always post questions like this in the facebook group...might be useful for all.

closed 2019-05-05 by tomvandeneede

paul999, 2019-05-05:
Yep, that answers it, thanks a lot for your work! Will post next time on the FB group :)