tomvandeneede/p2pp #21: Any tweaks needed for PrusaSlicer edition?

iandouglas, 2019-05-29:
I'm curious whether any changes are needed for working with the new, re-branded PrusaSlicer version, or if you're still evaluating that since it's pretty new.

tomvandeneede, 2019-05-29:
Ian,

as far as I know PrusaSlicer is a rename/rebranding of Slic3r without major modifications.  As long as the Multi material processing does not change significantly p2pp should work.  Quite a few people i n the community have tried it successfully so I think we are in the clear with PrusaSlicer.

iandouglas, 2019-05-29:
As far as I can tell, other than some UI elements moving around and looking slightly different, everything matches up. The only thing I couldn't find from your README under https://github.com/tomvandeneede/p2pp#filament-settings-

The filament settings tab in PrusaSlicer doesn't seem to expose linear advance right now, but the README is also a little unclear on setting a "greater than 0" value (but not suggesting what to set), or that "no action is required" in which case perhaps a preface saying "this is just FYI, you may not need to change this".

Houdini7, 2019-05-29:
Linear advance is a setting in the Custom gCode for the filament.
I believe it will be a “K” value.
Likely it will be a long expression/formula looking line (likely the 1st and only line)

tomvandeneede, 2019-05-29:
K-value is set to 0 when loading unloading filament by the Slic3r code.   This behaviour is used to detect state and end of the purge/filament swap sequence.   Setting the value back to something larger than 0 is an indication that purging has ended.   This is how P2PP currently works.   I am not aware of PrusaSplicer handling this differently from Slicar.. if so it would need further investigation

Houdini7, 2019-05-29:
I will absolutely defer to you on this. I’m rather new to gCode in general :)

I will say that many of us who have Bondtech extrudes installed had to set Linear Advance to 0 to stop the horrendous squeaking that Prusa’s current implementation of LA was generating.
I know Chris W. In the Facebook group has been working with several folks to get a better LA implementation into the firmware, and has recommended that we now use a value < 1. (Close to 0.6 I believe)

So it may be worth noting that P2PP needs a non 0 value to behave correctly?

jessegrunert, 2019-06-05:
I'd love to know if this working for PrusaSlicer as well, I'm very eager to get away from Canvas.

tomvandeneede, 2019-09-03:
Looks all is working with prusaslicer up to 2.1.0 with latest development release.
Master release up to 2.0.0

closed 2019-09-03 by tomvandeneede