tomvandeneede/p2pp #23: Strange Warning about Minimum Slice Length

Dragon2781, 2019-06-02:
Hi,
I just followed the installation instructions and want to print this Parrot from Thingiverse:
https://www.thingiverse.com/thing:2725110

I use the latest version of Prusa Slicer v2.0.0 and Python 3.7.3.

After I export the gcode the black p2pp script box comes up but then I get a warning about minimum slice length.

https://i.ibb.co/vZBvmrS/Warning.png

Even when I set the value ;P2PP MINSPLICE=60 I get the warning.

https://i.ibb.co/fd8FMNH/Warning2.png

What is goning wrong here? Please help.

tomvandeneede, 2019-06-02:
MINSPLICE is limited to a lower value of 70 because of hardware warranty at this moment
If you get short splices you have to increase the purge length until the splices meet the requirements.   If you have 1 or 2 short splices of nearly 70mm it should not be a problem, if you have too  many you may risk depleting the buffer if printing too fast.

Dragon2781, 2019-06-02:
Hi tomvandeneede,

do you mean the values of 'Unloaded Old filament" + "Loaded New Filament' in the small window?

tomvandeneede, 2019-06-02:
yes

> On 2 Jun 2019, at 14:48, Dragon2781 <notifications@github.com> wrote:
> 
> lues of 'Unloaded Old filament" + "Loaded New Filament' in the small window?
> 
>

Dragon2781, 2019-06-02:
ok I did this workaround.
Is it normal that I only get the **.gcode** file?
I thought that the Palette 2 needs a **.mcf.gcode** file???

tomvandeneede, 2019-06-02:
.mcf is only used by Canvas plugin to display the PALETTE logo next to the file name in the list. there is no other purpose.  It will zork without the mcd in the file name
If you want you can update the output file name format in the same place where you set the name of the script.

Dragon2781, 2019-06-02:
Thank you for your answers. Is there a forum or something for support for settings etc. ?
I have to wait for the actual print to finish before I try the p2pp again. Or should I post here?

tomvandeneede, 2019-06-02:
Bets is to post on the facebook group, we will add you then to the P2PP help talk where you can ask all questions to the experts

Dragon2781, 2019-06-02:
I can not find a p2pp group on FB. Can you please give me the link???

tomvandeneede, 2019-06-02:
we work from the 
Mosaic Palette, Palette+, Palette 2 (Pro) Users <https://www.facebook.com/groups/mosaic.users/?ref=group_header>

group

closed 2019-09-03 by tomvandeneede