tomvandeneede/p2pp #24: Filament color warning

jessegrunert, 2019-06-18:
I keep getting this warning in the output screen. I've tried adding the suggested code to both the start and end gcode sections of the filament config in PrusaSlicer. Am I missing something obvious? or is this a bug? Also, when I'm asked to load filament into the Palette, it says "please load black filament" for each of the inputs.

jessegrunert, 2019-06-18:
Filament #{} is missing Color info, make sure to add;P2PP FC=[extruder_colour] to filament GCode

tomvandeneede, 2019-06-19:
Can you please send me the code file so I can take a look at your config.
This looks like something is off in what I get from prusaSlicer.   Normally cools are encoded there and nothing else is neede

jessegrunert, 2019-06-28:
So it turns out I'm just an idiot. I typed "color" not "colour".

closed 2019-06-29 by jessegrunert