tomvandeneede/p2pp #25: P2PP Making Random blob at start of print

jeffreyso, 2019-06-29:
For some reason, at the start of the print going from the purge tower to the print, there’s somewhere where it adds a blob to the beginning which causes issues. I can’t seem to find where in the gcode but it seems that it doesn’t retract and keeps the filament moving as it comes out. Also the purge block seems to be printing really odd when the gcode is generated. It’ll do half of the first colour then the second half is printed in later layers which causes it to be partially detached from the buildplate. Using prusaslicer 2.00 not sure if older versions had better luck.

tomvandeneede, 2019-06-29:
Can you PM me the gcode on Facebook?  Curious what this may be but guess it is in your startup code
which profiles do you use?

jeffreyso, 2019-06-29:
Hi yes, what is your Facebook I will add you.

tomvandeneede, 2019-06-29:
easiest is to look fo r tom van den eede in the mosaic mfg moderators.

closed 2019-09-03 by tomvandeneede