tomvandeneede/p2pp #26: PURGETOWERDELTA use appears to lower during moves?

mmontminy, 2019-08-25:
I just used PURGETOWERDELTA for the first time and it appeared to start to lower Z while moving towards the purge tower, causing my cooling shroud to hit the print and knock over supports.

Is this expected behavior? I would assume the change in height wouldn't happen until the extruder is over the purge tower.
I had made sure to leave a considerable distance between the print and tower. The print was located at the far left-front of the bed, the tower at the rear-right.

skellatore, 2019-08-26:
Please give us a copy of your processed gcocde (processed by p2pp) and your unprocessed gcode.

you can use a service like wetransfer.com and select the “URL LINK” option rather than email.

Best Regards,
Tim Brookman

tomvandeneede, 2019-08-26:
is this with the development version or with the master version?

normally the z-move is issued as a separate command once the extruder is in the tower

please send us your gcode as tim suggested to take a look

tomvandeneede, 2019-08-28:
> Something I considered while thinking about what's happening, it occurs to me that the problem might be caused by "wipe to infill" in conjunction with PURGETOWERDELTA. Based on the slicer preview, and what it was doing, I believe it was performing a purge.
>
> It's as if the reduction in tower layers isn't taking into account the filament being purged into the object's infill and since purging was happening, p2pp assumed it was over the tower already. That would explain it only printing a single extrusion line in the purge block. I believe in one of the earlier failures, it started printing the next color shortly after the crash.

Looks like indeed there is a line commented out that should not have been. 
The issue appears to be in the G1 line that makes the move to the tower that is removed while it should not be 

G1 E-0.04000 F1800.00000 
; P2PP removed [Tower Delta] G1 X170.250 Y184.050 F7200.000 . <<<< this line 
G1 Z4.800 
G1 E0.80000 F1800.00000 

Could you please try the dev version of P2PP as quite a few fixes have been made, but cannot confirm for 100% your issue will be solved

tomvandeneede, 2019-08-28:
> So far the dev version is behaving. Wipe to infill is on, PURGETOWERDELTA set to the default 10, it's done a few tower moves, no crashes.

Thanks for the update

tomvandeneede, 2019-09-01:
Would it be possible to send me the unprocessed file?
This would give me something to simulate the problem with

mmontminy, 2019-09-01:
This is the 3mf I used (at least I think it is, I'm not terribly good about managing my projects).
[Bulbasaur.zip](https://github.com/tomvandeneede/p2pp/files/3563703/Bulbasaur.zip)

tomvandeneede, 2019-09-01:
Looks like latest dev solves the issue.  Could you please have a look?

tomvandeneede, 2019-09-01:
here 's some clarification:
G1 Z7.6 F10800
; CP TOOLCHANGE END
;------------------

G1  X233.227 Y180.429  F7200   << THIS IS THE MOVE IN THE TOWER BEFORE  GOING DOWN
   this line appeares to have been wrongly commented out in a previous release.
;--------------------
;P2PP Set wipe speed to 33mm/s
G1 F2000
G1 Z2.0 F10800    <<GOING DOWN
; CP EMPTY GRID START
; layer #38
G1  X174.227  E2.2424 F2900
G1  Y135.629  E1.7027
G1  X233.227  E2.2424
G1  Y180.429  E1.7027
G1  X174.227 
G1  Y135.629 
G1  X175.227
G1  Y180.429  E1.7027
G1  X184.727
G1  Y135.629  E1.7027
G1  X194.227
G1  Y180.429  E1.7027
G1  X203.727
G1  Y135.629  E1.7027
G1  X213.227
G1  Y180.429  E1.7027
G1  X222.727
G1  Y135.629  E1.7027
G1  X232.227
G1  Y180.429  E1.7027
G1  X175.227  F7200
G1 Z7.6 F10800  << GOING UP AGAIN BEFORE MOVING BACK TO THE PRINT
; CP EMPTY GRID END
;------------------

tomvandeneede, 2019-09-03:
Issue does no longer seem to appear with the latest version of P2PP

closed 2019-09-03 by tomvandeneede