tomvandeneede/p2pp #27: A few small things

tzabel, 2019-08-28:
In the current dev branch:
mcf.py line 302 and line 344 use single quotes instead of double quotes which results in ';--- P2PP removed [Tool Change]' and ';--- P2PP removed [Tower Delta] - ' to be interpreted as a single character and printing only ';' instead of the full string. 
A fix would be:
line 302: `        v.processed_gcode.append(";--- P2PP removed [Tool Change]" + gcode_full_line + "\n")`
line 344: `        v.processed_gcode.append(";--- P2PP removed [Tower Delta] - " + gcode_full_line + "\n")`

omega.py line 103 expects gui.ask_yes_no to return a boolean, but gui.ask_yes_no returns "YES" or "NO" (or with a different tcl installation "yes" or "no"), which is a non-zero string in both cases and will always be "true" when intrepreted as a boolean. 
A fix would be:

```
        if v.gui:
            result = gui.ask_yes_no('Not a Multi-Colour file?',
                              "This doesn't look like a multi-colour file. Skip processing?")
            if result == "yes" or result == "YES":
                exit(1)
```

tomvandeneede, 2019-08-28:
Thanks for reporting 

on the firs two I was thought that python allowed both single and double quotes for strings. I agree though it would be better practice to use the same way of representing strings every time. 

About the yes/no thsi is a known issue but it has net yet been fixed in the repository (still performing some tests on the new interface)

tomvandeneede, 2019-08-28:
These issues will be addressed in the next release

closed 2019-08-28 by tomvandeneede

tzabel, 2019-08-30:
Thanks for the fast answer and the development of this tool. We are using your tool/code as a base to use a palette 2 pro with our own developed 5-axis printer in accessory mode (which i know is still in development). You are right about the strings in python, I got confused coming from a C/C++ background. 
Two more things I found in the dev version:
- In accessory mode Ping pause 2 expects the absolute distance up to that point, not the relative distance to Ping pause 1. 
Possible fixes could be:
 1. `hexify_float(v.ping_extruder_position[i] + v.ping_extrusion_between_pause[i])` for the second argument in omega.py when writing the O31 command in accessory mode.
2. reworking v.ping_extrusion_between_pause to hold the absolute value instead of the relative value.

- A small thing in P2PP.py the variable v.consolewait is set after mcf.generate(), which causes it to do nothing.
A fix would be setting v.consolewait before calling mcf.generate().

tomvandeneede, 2019-08-30:
will check the ping dist but as far  as I remeber all distances are absolute... will double check...

tomvandeneede, 2019-08-30:
consolewait is going to be phased out with the new interface... will be gone in next release