tomvandeneede/p2pp #28: PURGETOWERDELTA seems to comment out everything except the tower

ianmcmahon, 2019-08-29:
Trying to use delta, and after laying down the first solid layer of the tower, it just moves back and forth between two points in the tower and overextrudes.  It looks like the moves for the model are being commented out.

master, delta set to 0 produces expected results:
https://gist.github.com/ianmcmahon/4ba9c714f05a86aabf616e6c2216e911

master, delta set to 50 removes the model:
https://gist.github.com/ianmcmahon/4893c39b30c7e554adf8f292501e3ccf

dev, delta set to 10 removes the model:
https://gist.github.com/ianmcmahon/66fd8daa9fc743eebddb812ffd661918

tomvandeneede, 2019-08-29:
from what I can see you do not have M900 Kn with n>0 in your filament start code as per documentation: 
; start_filament_gcode = "M572 D0 S0.05\n";"M572 D0 S0.05\n";"M572 D0 S0.05\n";"M572 D0 S0.05\n" 

this code is used as a delimiter in the gcode scanner of P2PP. Missing this delimiter, the processor will just continue to go back in the file and comment out code. 
If you have problems using M900 with your printer, you can just insert the following line in your filament gcode which will have the same effect. 
as you are using repap settings though; P2PP will process the M900 Kn command and comment it out afterwards 

I will make sure to add an extra chech to P2PP to raise a warning on this condition in the future. 

;P2PP ENDPURGETOWER 
can you please try this and report back if the issue still persists. 

[ https://gist.github.com/ianmcmahon/4ba9c714f05a86aabf616e6c2216e911 | https://gist.github.com/ianmcmahon/4ba9c714f05a86aabf616e6c2216e911 ] 

[ https://gist.github.com/ianmcmahon/4893c39b30c7e554adf8f292501e3ccf | https://gist.github.com/ianmcmahon/4893c39b30c7e554adf8f292501e3ccf ] 

[ https://gist.github.com/ianmcmahon/66fd8daa9fc743eebddb812ffd661918 | https://gist.github.com/ianmcmahon/66fd8daa9fc743eebddb812ffd661918 ]

ianmcmahon, 2019-08-29:
ok, I tried a few things and still wasn't able to get good results.

I added an M900 to my filament start.

I noticed that if I use REPRAPCOMPATIBLE, it will comment out the M900s, but that breaks the retrocorrect.

This file was done with the M900s in place, with ENDPURGETOWER.  It gets through the first layer and color change, and then on second layer it just stays at back left of the tower and moves upward with small extrusions, builds a little point, and then goes and tries to print the second layer of the model a few mm above the first layer.  I get similar results without ENDPURGETOWER.

https://gist.github.com/ianmcmahon/557d02f9e4cd1ced1b9a372789876786

tomvandeneede, 2019-08-29:
will look into it
please from your side check of all setup has been done as described.

ianmcmahon, 2019-08-30:
I believe I have done all setup.  Everything is working perfectly without tower delta.

I've been through the code a bit trying to understand what's causing the issue, unfortunately I've had a hard time understanding the intent.  It seems like it should be commenting all moves between EMPTY GRID START and EMPTY GRID END, but instead, upon hitting EMPTY GRID START, it seems that it calls retrocorrect and then works backwards, commenting out everything before the empty grid, until it hits M900?   If I could understand a bit more about why it's written this way, I may be able to troubleshoot.

tomvandeneede, 2019-08-30:
that’s more or less it... will try to have a look over the weekend.  new dev version should solve it

with the new purge tower and side wipe code has becomen increasingly complex and position dependent.  maybe I will try a rewrite of some parts whan palette plus ans accessorynmode functions are all fully sorted out.   Dont worry about the solutions... the better problem descriptions I get the easier
to be honest I dont have any means of testing reprap so good to know somebody with proper testing capability

ramble392, 2019-08-30:
I seem to be having a similar issue with the PURGETOWERDELTA settings. In my case I have a model which has the first 20 layers or so with no filament changes. What happens is the tower is built correctly, but the model is missing the bottom layers until the first filament change.  I'm using the V3.3.3 beta edition of P2PP along with the Prusa Slicer 2.1.0 Beta2 edition.

I've tried the above suggestions and it doesn't seem to change the output.

[image]

tomvandeneede, 2019-08-31:
please send me your gcode file

ramble392, 2019-08-31:
Ok, I've attached a zip for the unprocessed and processed versions of the gcode. This one uses the M900 Kn command in the  start filament gcode.

[HedgeHog.zip](https://github.com/tomvandeneede/p2pp/files/3562537/HedgeHog.zip)

tomvandeneede, 2019-08-31:
Looks like a discrepany in the reprap code.  I have some idea to fix this without even needing the M900 anymore.  Will see if I can get this implemented tomorrow

tomvandeneede, 2019-09-01:
Issue is related to changes in PrusaSlicer 2.1.0 
3.3.4 should solve these issues.  
This version is currently in dev mode, please report any new/similar problems as separate issues

closed 2019-09-01 by tomvandeneede