tomvandeneede/p2pp #30: 3.3.7 has strange move at beginning of toolchange wipe

ianmcmahon, 2019-09-04:
Thanks Tom, I pulled 3.3.4, and the tower delta is working as expected for me now, but there were issues in it that caused the splice and ping locations to be wrong.  I see that 3.3.7 seems to fix these issues, running a calibration print with it now.

I've noticed a minor issue with 3.3.7, at the beginning of toolchange wipe, it makes a rapid move and extrude to the opposite side of the tower, leaving a big blob.

Here is 3.2.1's output:
`2035 ;--- P2PP removed [Tool Change]T0
 2036
 2037 ;--- P2PP removed [RepRap Processing] -M900 K0.05
 2038 M572 D0 S0.05
 2039
 2040 G4 S0
 2041 G1  Y142.400
 2042 G0 X230.082 Y142.4
 2043 ; CP TOOLCHANGE WIPE
 2044 G1  X181.000 E1.8654 F1600
 2045 G1  X180.250 Y142.900  E0.0343`

and here is 3.3.7's output:
`2038 ;--- P2PP removed [Tool Change]T0
 2039
 2040 ;--- P2PP removed [RepRap Processing] -M900 K0.05
 2041 M572 D0 S0.05
 2042
 2043 G4 S0
 2044 G1  Y142.400
 2045 G1 Z0.40 F10800
 2046 ; --- P2PP Set wipe speed to 2000.0mm/s
 2047 G1 F4000.0
 2048 G0 X180.25 Y142.4
 2049 ; CP TOOLCHANGE WIPE
 2050 G1  X181.000 E1.8654 F1600
 2051 G1  X180.250 Y142.900  E0.0343`

Notice how the `G0` just before `; CP TOOLCHANGE WIPE`  in 3.3.7 goes to 180 instead of 230?  Then it makes a 1.86mm extrusion while moving less than a mm, causing a blob:

[image]

ianmcmahon, 2019-09-04:
Also 3.3.7 is commenting out the shutdown gcode at the end of the print.

3.2.1:
```
26497 ; CP TOOLCHANGE END
26498 ;------------------
26499
26500
26501
26502 G1 E-0.80000 F1800.00000
26503 G1 Z7.850 F9000.000
26504 M107
26505 ; Filament-specific end gcode
26506 M106 S0
26507 M104 S0
26508 M140 S0
26509 G91
26510 G1 E-1.0 Z+10.0 F1000.0
26511 G90
26512 G1 F6000 X10 Y280
26513 M84
26514 ; filament used [mm] = 1900.2, 866.6
```

3.3.7:
```
26539 ; CP TOOLCHANGE END
26540 ;------------------
26541
26542
26543
26544 ;--- P2PP removed [Tool Unload]G1 E-0.80000 F1800.00000
26545 ;--- P2PP removed [Tool Unload]G1 Z7.850 F9000.000
26546 ;--- P2PP removed [Tool Unload]M107
26547 ; Filament-specific end gcode
26548 ;--- P2PP removed [Tool Unload]M106 S0
26549 ;--- P2PP removed [Tool Unload]M104 S0
26550 ;--- P2PP removed [Tool Unload]M140 S0
26551 ;--- P2PP removed [Tool Unload]G91
26552 ;--- P2PP removed [Tool Unload]G1 E-1.0 Z+10.0 F1000.0
26553 ;--- P2PP removed [Tool Unload]G90
26554 ;--- P2PP removed [Tool Unload]G1 F6000 X10 Y280
26555 ;--- P2PP removed [Tool Unload]M84
26556 ; filament used [mm] = 1900.2, 866.6
```

tomvandeneede, 2019-09-04:
can you send me the unprocessed gcode file please

ianmcmahon, 2019-09-04:
[keychain-unprocessed.gcode.txt](https://github.com/tomvandeneede/p2pp/files/3575566/keychain-unprocessed.gcode.txt)
[keychain-3.3.7.gcode.txt](https://github.com/tomvandeneede/p2pp/files/3575567/keychain-3.3.7.gcode.txt)

tomvandeneede, 2019-09-04:
End code should be solved in latest version 3.3.8
looking into other issue

tomvandeneede, 2019-09-04:
3.3.8 should fix both problems

ianmcmahon, 2019-09-04:
Confirmed, thanks.  There's another oddity, it's leaving a blank space in the first few rows of the tower.  I'm closing this issue, and I'll do some research, and open a new issue when I have some information gathered for you.

[image]

closed 2019-09-04 by ianmcmahon

tomvandeneede, 2019-09-04:
this is normal, this is an unneeded unload sequence part on the MMU that is removed.  In future release I am planning to close this gap.
should be there somewhere next week

ianmcmahon, 2019-09-04:
Ahh ok, thanks.  It was present in earlier versions, so I was worried it was a regression.  As long as it's accounted properly in the splice budget, it's not a concern to me! :D

tomvandeneede, 2019-09-04:
did a special tool for the splice budget check and all comes down to 0.001mm precision on a per ping basis so should be close

ianmcmahon, 2019-09-04:
oh nice, I'd love to see that tool!

On a related note, I'm thinking about writing a tool that will generate gcode to print a purge volume test, basically print a bunch of squares in a single layer, each of which is a purge block, covering all the filament combinations currently loaded.    Like say 400mm3 worth of purge, with lines printed alongside to indicate every 50mm3 of purge, so you can quickly eyeball how much purge volume you need for a set of colors.

tomvandeneede, 2019-09-04:
Looks like a fun idea ans fairly easy to make I guess

Only issue is that you need to have the color changes spot on.  This is not always easy as the first layer tends to be the one with most problems due to Z-height etc… but this will make it even more of a challenge.

 

What language are you thinking of using for this?

 

 

Also keep in mind start and stop code that needs to be configurable depending on the printer… 

 

I am working on condensing the space in the putge towers as w next step and just sharing the challenges I see when looking into this… don’t let this make you change ideas about making this :)

 

You could actually do this as a combination of an stl and  use prusaslicer to generate a one color gcode file that you color using a python or perl scrupt like P2PP.

Should be plain fun to write and keep operational and should help people find proper transition settings more easily

ianmcmahon, 2019-09-05:
I got the beginnings of a purge test tool written:

https://github.com/ianmcmahon/purgetest

It's written in go, it takes an output gcode from prusaslicer and uses it as a config file, and generates gcode from scratch with splices and pings, and prints a 500mm3 purge square for every transition combination:

[image]

tomvandeneede, 2019-09-05:
nice work

tomvandeneede, 2019-09-05:
what did you write this in?

ianmcmahon, 2019-09-05:
in Go, my language of choice by a longshot.

I just got done writing a splice budget tool also, which sums up all the E moves and inserts comments showing where the splices should fall.  Based on that, P2PP is behaving correctly, all splices fall at TOOLCHANGE WIPE + SPLICEOFFSET +/-2mm.

My purge generator also shows the behavior of early transitions, getting earlier as the print progresses.  I'm going to play with loading offset to see if that helps.

tomvandeneede, 2019-09-05:
I use a similar program to check my code (written in Py)

go seems to be a nice language.  maybe I need to give that a try
not sure though how you can write a portable user interface in there without going into the depths of OpenGL

would be nice to do some C++ though

Did you already do a code parser to inspect gcode in Go?

ianmcmahon, 2019-09-05:
There's ways to do portable UIs, but I mostly avoid native UI.  Go is perfectly suited to systems work and internet glue code.

I don't have anything like a full featured parser, I just did a handful of regexes to accomplish specific things.

tomvandeneede, 2019-09-05:
maybe rewriting p2pp in go would not be a bed thing… would take just a little time I guess but for the benefit of having a super easy installation

ianmcmahon, 2019-09-05:
I'll put some time into a proof of concept.  I was able to figure out most of the mcf stuff from looking at your code, the only thing I'm unclear about at this point is the "splice algorithms" (O32).

Also I'm curious about the hotswap code (O29).  Is that something that'll instruct the p2 to pause and prompt for a color change?  That's something I want!

tomvandeneede, 2019-09-05:
MCF is fairly straightforward, but grown a lot in complexity over the past few months as features were added

yep, do not know how it works either
all the rest is easy

do you know if hot swap works canvas?  In prusaslicer we could easily add a more colors and do the most optimal swap based on the upcoming colors
would not eaven be difficult to do

what development side do you use for Go.  Looked at goLang, but seems very expensive for what it does (looks neat though)

ianmcmahon, 2019-09-05:
I just use vim, with some plugins that are wrapped up nicely here:
https://github.com/farazdagi/vim-go-ide

I don't know anything about canvas, I tried it briefly when I first got my P2, but I've put so much time and effort into understanding and tuning slic3r that I went straight to p2pp.  If I get some time, I'll play with it a bit and see if it will generate hotswaps

tomvandeneede, 2019-09-05:
Ian,

 

Would you mind sending me your email address (t.vandeneede@pandora.be <mailto:t.vandeneede@pandora.be> ) so we can talk outside of the issue list

 

https://github.com/farazdagi/vim-go-ide