tomvandeneede/p2pp #31: Double Z-hop on layer change when skipping tower

ianmcmahon, 2019-09-04:
3.3.8 is printing about perfectly, one last minor issue:

There's a Z-hop moving to the tower that doesn't get commented out, so I get a double Z-hop on layer change, which leaves a small pip.

```
 4623 ;LAYER 4
 4624
 4625 G1 Z1.000 F9000.000
 4626 G1 E-10.00000 F1800.00000
 4627 G1 Z1.250 F9000.000
 4628 ;--- P2PP removed [Tower Delta] - G1 X180.500 Y140.500
 4629 G1 Z1.000
 4630 G1 E10.00000 F1800.00000
 4631 ;--------------------
 4632 ; CP EMPTY GRID START
 4633 ;--- P2PP removed [Tower Delta] - ; layer #5
 4634 ;--- P2PP removed [Tower Delta] - G1  X180.000 Y140.500
 4635 ;--- P2PP removed [Tower Delta] - G1  Y140.000
```

[keychain-unprocessed.gcode.txt](https://github.com/tomvandeneede/p2pp/files/3575754/keychain-unprocessed.gcode.txt)

tomvandeneede, 2019-09-04:
will be something for tomorrow but I will look into it.

thanks for reporting

ianmcmahon, 2019-09-04:
one other thing, in this file there's no toolchange on the first layer, so it does a full purge layer to make the bottom of the tower.  P2PP is commenting out the feedrate in the purge, so it's retaining the Z move feedrate of 10800mm/m.  It does the same thing on the second layer, where there's an unskipped empty grid.

[head-unprocessed.gcode.txt](https://github.com/tomvandeneede/p2pp/files/3575843/head-unprocessed.gcode.txt)

```
   399 ; TOWER COORDINATES (  176.17,  136.17) to (  245.00,  209.53)
   400 ; CP WIPE TOWER FIRST LAYER BRIM END
   401 ;-----------------------------------
   402 G1  X239.500 Y140.500  F7200
   403 ;--------------------
   404 ; --- P2PP Set wipe speed to 66.7mm/s
   405 G1 F4000.0
   406 G1 Z0.20 F10800
   407 ; CP EMPTY GRID START
   408 ; layer #1
   409 G1 X240.000
   410 G1 Y140.000
   411 G1 Y205.700 E2.4970 ;--- P2PP Removed [Removed Parameters] - F1450
   412 G1 X180.000 E2.2804
   413 G1 Y140.000 E2.4970
```

tomvandeneede, 2019-09-04:
Hmm that’s new as well.  Will look into that as well tomorrow

Just resumed work after a 3 week vacation so things are hectic specially with the changes in Prusa playing up

If I have time I will look into this in the morning

Should insert a speed setting by default… not sure why it is not doing that

tomvandeneede, 2019-09-05:
Should be ok now

;--- P2PP removed [RepRap Processing] -M900 K0.05
M572 D0 S0.05

G4 S0
G1  Y152.000
G1 Z0.20 F10800
; --- P2PP Set wipe speed to 33.3mm/s
G1 F2000
G0 X189.918 Y152.0
; CP TOOLCHANGE WIPE
G1  X239.000 E2.2012 F800

ianmcmahon, 2019-09-05:
ok great, I'll pull and check it out as soon as this print finishes.

There's another thing, when it's moving down in Z to the tower, I get a little extrude right before the plunge that turns into a dangly string on the side of my purge block. I hunted for the section of gcode that's got a spurious extrude move, but I can't find it

I'm also struggling with getting good prints, because there's consistently a 95% first ping and ~98% second ping, I get a lot of early transitions until ping stabilizes at 100+/-0.5%.  I bumped spliceoffset up to 60mm which is getting me through a print, but it seems like a bad solution.

tomvandeneede, 2019-09-05:
low first ping could indicate Z-height problem where the nozzle gets more resistance during this layer and the back pressure reduced the amount of extrusion.
I will double check counts on the first layer and see if I can find anything abnormal there.

I will check into the extrude as well…. Should be no problem to add 0.4mm of retract /unretract around the Z-move if needed

tomvandeneede, 2019-09-05:
;--- P2PP removed [Tool Change]T1

;--- P2PP removed [RepRap Processing] -M900 K0.05
M572 D0 S0.05

G4 S0
G1  Y152.000
G1 E-0.4
G1 Z0.20 F10800
G1 E0.4
; --- P2PP Set wipe speed to 33.3mm/s
G1 F2000
G0 X189.918 Y152.0
; CP TOOLCHANGE WIPE

done

ianmcmahon, 2019-09-05:
yeah see here's where I'm confused... low ping means it's underconsuming filament, yes? If it's underconsuming, I would expect a late transition, but I get an early transition.

ianmcmahon, 2019-09-05:
the file above `head-unprocessed.gcode` is the print in particular I'm talking about.   The first transition happens 10mm up.  Before I increased the splice offset to 60, the first transition would be over a layer early:
[image]

I'm printing a model right now that's just two small cubes of different color, no tower delta at all, a transition on every layer.  It's printing perfectly, transitioning perfectly.  First ping was 98.4%.

ianmcmahon, 2019-09-05:
So two side by side squares with no purge delta printed perfectly.

I created this model to test something similar, but with 2.4mm of purge delta before the first transition to white, and it transitions early.

[image]

closed 2019-09-06 by tomvandeneede