tomvandeneede/p2pp #32: Dev version removed custom tool change code, along with temp M106

chernogorsky, 2019-09-08:
If you try working with different temp filament, it does not work properly

it has
;--- P2PP removed [Tool Unload]M104 S210 
instead of 
M104 S210

tomvandeneede, 2019-09-08:
This is being worked on

Have found some other issues with the side wipe

 

Rewrite is taking a little longer than expected

chernogorsky, 2019-09-08:
That one with a regular purge tower, not a side. 
PrusaSlicer 2.1.0.rc+20190904

chernogorsky, 2019-09-08:
[support_test_low.3mf.txt](https://github.com/tomvandeneede/p2pp/files/3587957/support_test_low.3mf.txt)

tomvandeneede, 2019-09-09:
yep I understand
but currently there is also an issue with sidewise I am fixing, this is why I cannot immediately fix your issue, but it will be part of the current fix

tomvandeneede, 2019-09-11:
I uploaded 3.4.0 that I expect should solve this issue.
Still beta but worth the try

tomvandeneede, 2019-09-13:
Latest version does no longer remove any commands related to temperature

closed 2019-09-13 by tomvandeneede