tomvandeneede/p2pp #34: Win 10 / PS 2.0 / p2pp dev (16.09.2019)

geka79, 2019-09-16:
I get the following error message after the graphic progress window:

Traceback (most recent call last):
  File "C:\......\p2pp-dev\\p2pp.py", line 145, in <module>
    main(vars(arguments.parse_args()))
  File "C:\.......\p2pp-dev\\p2pp.py", line 91, in main
    args['silent']
  File "C:\.......\p2pp-dev\p2pp\mcf.py", line 587, in generate
    parse_gcode()
  File "C:\.......\p2pp-dev\p2pp\mcf.py", line 290, in parse_gcode
    if v.parsedgcode[-1].get_parameter("E", 0) < 0:
TypeError: '<' not supported between instances of 'str' and 'int'
Drücken Sie eine beliebige Taste . . .

tomvandeneede, 2019-09-17:
could you please send me the gode file you tried to process?

geka79, 2019-09-17:
No Problem. I will send it in the evening.

Gesendet mit der GMX iPhone App

geka79, 2019-09-17:
Gesendet: Dienstag, 17. September 2019 um 07:19 Uhr

tomvandeneede, 2019-09-18:
issue was with M84 command with explicit callout of stepper axis info.
pasring has been updated to prevent this from happening

closed 2019-09-18 by tomvandeneede