tomvandeneede/p2pp #35: Wipe tower Slicer 2.1 and dev version of p2pp

audiorazor, 2019-09-23:
I’m have terrible issues with wipe towers generated in 2.1 slicer while using the dev version of p2pp.   When it switches from no infill to infill, the layers do not adhere and curl up.   Rolling the system back to 2.0 and master p2pp fixed the issue.  
[image]
[image]
[image]
[image]

tomvandeneede, 2019-09-23:
Hi,

thanks for reporting this issue
can you please give some further info on your settings or provide me with the unprocessed gcode of this file so I can take a look where the problem is situate.
Can you please also confirm the version of p2pp  you are working with for reference.
In the last two days some minor changes have been implemented that could resolve your issue

tomvandeneede, 2019-09-23:
I checked and there was no reraction around the wipe tower
I explicitly added it
Can you please confirm this solves your issue?

audiorazor, 2019-09-23:
Thanks Tom.  It will be a week before I can test it again but I will
message you as soon as I can.  Do you still want my files or think that you
figured it out?

tomvandeneede, 2019-09-23:
I hope I figured it out.  I tested the ayers and they seem to be in climbing order per 0.2 layer height progress as expected so I don’t think looking at your files will make me change st now

thanks for testing

audiorazor, 2019-09-23:
When I overwrite p2pp with the new version, do I need to keep the .bat file on a Mac?  or can I just overwrite the entire folder?

tomvandeneede, 2019-09-24:
on a mac you can just overwrite everything

audiorazor, 2019-09-30:
I just got home and am testing your updated dev files now.   I'll let you now how it goes.

audiorazor, 2019-09-30:
Nice work, it appears to be working just as well as the master branch now.   Thank you for fixing it so quickly.

closed 2019-09-30 by audiorazor