tomvandeneede/p2pp #36: stop extruding after a few layer

geka79, 2019-09-23:
ps2.0 /  p2pp master

at the lastprint i had the problem, that the Extruder stop working.

i had no idea, whats could the problem.

layerheight 1mm: stop extruding at the purge tower.
no blocked nozzle.

[Skunk_Eyes_0.2mm_11h10m.mcf.zip](https://github.com/tomvandeneede/p2pp/files/3644447/Skunk_Eyes_0.2mm_11h10m.mcf.zip)

tomvandeneede, 2019-09-24:
Checked the attached file and I can see no reason why it would stop at height 1 from an extrusion perspective
could it be that you had a buffer underpin on the p2 device itself?
This sometimes happens but as far as I know is not P2PP or gcode related other than you trying to print too fast

geka79, 2019-09-24:
At splicing, the federate reduce to 60%.
No ping/pong problem. All around 99-102%. 

At canvas sliced products all running smooth but very long. 

I will try to delete p2pp und ps2.0 from my laptop and reinstall. 

Gesendet mit der GMX iPhone App

tomvandeneede, 2019-10-16:
no further feedback

closed 2019-10-16 by tomvandeneede