tomvandeneede/p2pp #37: Purge Block Prusa Slicer 2.1

Crane49, 2019-10-16:
Any one know what is causing the extra extrusion in the purge block
[image]

tomvandeneede, 2019-10-16:
there seems to be a setting in ps causing this.  I try to filer out these lines while processing
this preview is NOT the final file though, prusa does not load the post-processed gcode file 
If you want you can load the file in an online model previewer and check if the lines are still there

I saw some recent activity in the Prusa slicer code that could be there to fix this so a net update might no longer have these lines

Crane49, 2019-10-20:
You are correct, this is a PrusaSlicer Issue as it didn't show up in my print at all. Thanks a lot for the plugin as it has made the Palette 2 a useful tool as it is almost totally automatic. I have 2 Mk3s one with MMU2s and one with the Palette 2.

Mike C

closed 2019-10-20 by Crane49