tomvandeneede/p2pp #39: Splice length warnings when using firmware retraction

cmroche, 2019-11-10:
My printer is setup to use firmware retractions in Prusa 2.1 (I'm working in the dev branch), it will generate short splice warnings for nearly all layers and the final report will have a total printed length much much shorter than it should be.

If I disable firmware retraction and setup to use retraction in Prusa directly everything appears to work correctly.

[Tests.zip](https://github.com/tomvandeneede/p2pp/files/3828774/Tests.zip)

tomvandeneede, 2019-11-10:
can you please send me also an unprocessed fie (file before p2pp runs) so I can simulate the behaviour of p2pp

tomvandeneede, 2019-11-10:
I have tried to get around the issue with 3.5.19 dev version.  Can you please check if this one behaves as it should?  if not please send me the unprocessed file from your print so I can take a look where it fails.

cmroche, 2019-11-10:
Sure thing. New set of tests, with unprocessed output using firmware retraction, as well as the processed output for the same models with and without firmware retraction.

[NewTest.zip](https://github.com/tomvandeneede/p2pp/files/3828897/NewTest.zip)

cmroche, 2019-11-10:
I've been testing a bug fix, I'll share it since it might help, it does appear to address the problem. With this I no longer get the warnings, and the final print length matches exactly the summary in Prusa Slicer now.

cmroche, 2019-11-10:
@tomvandeneede 3.5.19 appears to work correctly as well.

tomvandeneede, 2019-11-10:
If it is ok I’ll leave it at this fix… but will have a look at your changes and decide later.  I m waiting for the next prusa slicer version to appear to possibly get rid of a lt of this code to start with
Thanks for reporting and testing and suggesting a fix.
I never actually met someone using firmware retraction… so good to know this should be covered as well… May I ask what the advantage of these firmware retractions is over the regular ones… cannot be that bot otherwise everybody would be using them…

cmroche, 2019-11-10:
I use a Duet2 system, which gives me the ability to setup machine configurations per material/tool. Firmware retraction allows me easy changing of parameters between materials through the Duet Interface. It also gives me a simple mechanism to tune the Z-Hop or Retract parameters during a print.

A quick question though, is it sufficient only to scan for the G11 in backpass? Or should the G10 also be present?

The change that I sent didn't address this function as I had not found it yet. However it does bring support for tracking the prusa config and generating G10 and G11 commands for purge and sidewipe.

In any case, thanks for the quick response.

tomvandeneede, 2019-11-10:
including your code as we speak, some minor typos 
only issue I see is that retracts will no longer be part of the ping checks as you don’t have any actual value for them
should be minor but could add up to 0.3% or so…

tomvandeneede, 2019-11-10:
created 3.5.20, could you please confirm if this stil works.  It has your changes in it
I tested for some other files as well and seems to hold…

wel flor P2PP the G11 is sufficient.   It is a specific pattern of code I am looking for to determine the start of the unload sequence
G10 and G11 would be needed in case I wanted to account for filament counter changes in the P2 but as this can be tuned it is going to be disfunct anyway

if you confirm all is still working I’ll close the issue

and thanks again for your contribution

cmroche, 2019-11-10:
Haha you were not kidding about those typos. Sorry, really threw this together fast, thanks though. At first test it checks out. I'll setup for my next print tomorrow evening and confirm once it is done.

tomvandeneede, 2019-11-10:
super!

closed 2019-11-12 by cmroche