tomvandeneede/p2pp #44: First filament segment is too long, PS 2.1 and dev branch

MalikDrako, 2019-11-14:
After updating to PrusaSlicer 2.1 and switching to the dev branch (pulled yesterday), I'm getting a first filament segment that's too long by about 2m and it missed every single ping for that print. Prior to updating, it was working fine.

I found a method that should add up all the E movements and got numbers that were a few cm off from the numbers in the p2pp gcode file except for the first segment.

[Gcode.zip](https://github.com/tomvandeneede/p2pp/files/3848270/Gcode.zip)

tomvandeneede, 2019-11-14:
cn you please confirm the version of P2PP you are using.
I have received another report of this behaviour and this should now be solved in 3.5.26

> On 14 Nov 2019, at 20:24, MalikDrako <notifications@github.com> wrote:
> 
> After updating to PrusaSlicer 2.1 and switching to the dev branch (pulled yesterday), I'm getting a first filament segment that's too long by about 2m and it missed every single ping for that print. Prior to updating, it was working fine.
> 
> I found a method that should add up all th
>

MalikDrako, 2019-11-14:
Is it in version.py? If so, 3.5.24.
I'll try pulling the latest and see if it fixes the issue

MalikDrako, 2019-11-14:
The lengths look reasonable, I'll try running it once my printer finishes a print in about 2h

MalikDrako, 2019-11-15:
I'm guessing it's a slicer or physical issue (I was hearing what seemed like the extruder gear slipping), but I got a little bit of color bleed right before the first switch, and the transitions were off for the first 1.5mm or so. Above that, the colors were perfect. 
I run the Canvas Hub plugin for Octoprint and the ping values were 105% +- 0.5% for the whole print. The pong values were 99.8%-99.99%

tomvandeneede, 2019-11-15:
Could you please PM me the files.   Would like to take a look if all looks ok
Gear slipping and 105% are contradicting in a way… unless you are really over extruding more than you should
`Did you calibrate your extruder?

MalikDrako, 2019-11-15:
Ping values of > 100% mean the printer used more filament than expected, correct?
I haven't calibrated my extruder, but the MK3S is pretty close to correct from the factory (and previously the ping values were right around 100%)

Here are the files, I didn't think to save a copy of the gcode before it was run through p2pp, so I included the models, the processed file I printed, and before and after gcode for an almost identical platter.
[Clutch_Gcode.zip](https://github.com/tomvandeneede/p2pp/files/3852725/Clutch_Gcode.zip)

tomvandeneede, 2019-11-15:
100% means over extrusion
what numbers do you get?

tomvandeneede, 2019-11-15:
Calibration is very important as is first layer calibration
I will take a look at the speed as something is strange there.   The processed print prints at higher speed than the original so that should not be the case

tomvandeneede, 2019-12-20:
closed while missing info

closed 2019-12-20 by tomvandeneede

MalikDrako, 2019-12-21:
Did you ever find anything related to the speed?
I did check the extruder calibration and it was pretty much right on - 100mm filament commanded and about 100mm filament was extruded.

I updated to the latest master as of this post and am still seeing the issue. I observed the extruder gear and it does seem to jump slightly when I hear the sound I mentioned above. The strange thing is it seems to be happening at about the same rate regardless of extrusion speed, possibly at the start/end of a line segment. (I'm printing test cubes, so the segments are comparatively long)

MalikDrako, 2019-12-21:
I played with it a bit more and found what was causing the issue. Disabling `;P2PP ABSOLUTEEXTRUDER` stopped the skipping and the out of sync color changes. dd61fba74dfbb9f473a1a282db6b6febc5bd09b6 sounds like it might fix the issue, I'll give the dev branch a try next week to see.

Unprocessed and processed gcode with and without `;P2PP ABSOLUTEEXTRUDER` [cube_skip_test.zip](https://github.com/tomvandeneede/p2pp/files/3990925/cube_skip_test.zip)

tomvandeneede, 2019-12-21:
> I played with it a bit more and found what was causing the issue. Disabling ;P2PP ABSOLUTEEXTRUDER stopped the skipping and the out of sync color changes. dd61fba <https://github.com/tomvandeneede/p2pp/commit/dd61fba74dfbb9f473a1a282db6b6febc5bd09b6> sounds like it might fix the issue, I'll give the dev branch a try next week to see.

let me know.  I checked extrusion and looks fine in latest version.   Deviation is less then 0.005mmper pin.  tested only first few pings as labour intensive bit the process seems to be ok

MalikDrako, 2019-12-24:
I ran some more tests, it looks like the 105% pings were from extrusion multiplier compensation, although that doesn't explain why I hear the extruder skipping in absolute mode (even on the slower sections, seems like the end of segments).
Both 95% and 100% flow in absolute mode look over-extruded by about the same amount, and 95% in relative mode looks better

tomvandeneede, 2019-12-24:
will work on this once PS2.2 is stable.  No idea what the cause may be.  If you have forte further tests please send me som eunrpocessed and processed files to check