tomvandeneede/p2pp #45: Separate side purge filament feed rate from move rate

mmontminy, 2019-12-07:
Using the dev version, 3.6.6.

Is it possible to add another parameter to control movement separate from filament feed rate.
I end up using a fairly low feed rate as it seems to cause skipped steps if I'm just extruding into air if I feed too fast. 
The x/y movement while purging is very slow, as is the move back to the print.

tomvandeneede, 2019-12-07:
Can you please elaborate what this parameter should do?
I am not sure I follow for 100%

mmontminy, 2019-12-07:
Watching the print more and thinking about this more, there might be a bug...
I have P2PP WIPEFEEDRATE=500.
The first move to the purge area seemed ok, as did the purge itself. When it moved back to the print though, it did so at 1mm/s.
The next purge it moved back to the print at 6mm/s
It appears to alternate between 1mm/s and 6mm/s moves back from the purge area.
My travel speed is set to 150mm/s.

My initial though was to have a second parameter to define the speed to/from the purge area, but it seems like it should just consider that a travel, but it doesn't appear to.
```
;---------------------------
;  P2PP SIDE WIPE: 121.996mm
;P2PP BEFORESIDEWIPEGCODE ;--ENTER GCODE TO BE EXECUTED BEFORE SIDEWIPE (one coomand per line,can be multiple lines)
G1 F8640
G0 X-7.000 Y120.000
G1 E0.80000
G1 X-7.000 Y130.000 E20.00000 F33
G1 X-7.000 Y120.000 E20.00000 F33
G1 X-7.000 Y130.000 E20.00000 F33
G1 X-7.000 Y120.000 E20.00000 F33
G1 X-7.000 Y130.000 E20.00000 F33
G1 X-7.000 Y120.000 E20.00000 F33
G1 X-7.000 Y130.000 E1.99580 F334
M98 P"/macros/Nozzle Wipe.g"
G1 E-0.80000
;---------------------------
G1 X154.683 Y177.885
```
The final move seems to happen at the same rate as the last extrude.

tomvandeneede, 2019-12-07:
Did 3.6.7 with speedup on return

tomvandeneede, 2019-12-07:
---------------------------
;  P2PP SIDE WIPE: 155.790mm
G1 F8640
G0 X253.900 Y25.000
G1 E0.80000
G1 X253.900 Y175.000 E20.00000 F3000
G1 X253.900 Y25.000 E20.00000 F3000
G1 X253.900 Y175.000 E20.00000 F3000
G1 X253.900 Y25.000 E20.00000 F3000
G1 X253.900 Y175.000 E20.00000 F3000
G1 X253.900 Y25.000 E20.00000 F3000
G1 X253.900 Y175.000 E20.00000 F3000
G1 X253.900 Y25.000 E15.78980 F3799
G1 E-0.80000
G1 F8640
;—————————————

mmontminy, 2019-12-07:
That seems to be working much better.

tomvandeneede, 2019-12-07:
please let me know if all ok after your print finishes

happy printing….

mmontminy, 2019-12-07:
I did notice that it started off the print with a purge. I think I've seen this before so I don't know if it's new to this release.

The print starts, my normal prime routine happens, and instead of starting the print, it starts purging as though it's doing a color change. Then the print starts.

I've attached the gcode up to the start of print.

[p2pp-start.zip](https://github.com/tomvandeneede/p2pp/files/3935570/p2pp-start.zip)

tomvandeneede, 2019-12-07:
getting late here will look in the morning

tomvandeneede, 2019-12-08:
have a look at 3.6.8

mmontminy, 2019-12-08:
That seems to have fixed it. Both issues appear to be resolved.

closed 2019-12-20 by tomvandeneede