tomvandeneede/p2pp #47: Problems Processing with wipe tower enabled

mjgorman, 2019-12-09:
Slicer: PrusaSlicer-2.1.0 Linux version

I'm sure I must be doing something wrong since no one else is complaining, and this i my first time trying to get it setup and working after playing with canvas for the last few days since receiving my P2S Pro.

If i run with Wipe Tower enabled as directed by the README and its moved off the bed to enable side purge i get this error:

```
Post-processing script /home/mgorman/.pyenv/shims/python /home/mgorman/sites/p2pp/P2PP.py -i on file /tmp/.PrusaSlicer.upload.d072-ecac-3e1c-a1f2 failed.
Error code: 1
Output:
Traceback (most recent call last):
File "/home/mgorman/sites/p2pp/P2PP.py", line 145, in <module>
main(vars(arguments.parse_args()))
File "/home/mgorman/sites/p2pp/P2PP.py", line 91, in main
args['silent']
File "/home/mgorman/sites/p2pp/p2pp/mcf.py", line 827, in generate
convert_to_absolute()
File "/home/mgorman/sites/p2pp/p2pp/mcf.py", line 72, in convert_to_absolute
to_e = float(fields[j][1:])
ValueError: could not convert string to float: 
```

If i put the tower back in the bed I get:
```
Post-processing script /home/mgorman/.pyenv/shims/python /home/mgorman/sites/p2pp/P2PP.py -i on file /tmp/.PrusaSlicer.upload.d072-ecac-3e1c-a1f2 failed.
Error code: 1
Output:
Traceback (most recent call last):
File "/home/mgorman/sites/p2pp/P2PP.py", line 145, in <module>
main(vars(arguments.parse_args()))
File "/home/mgorman/sites/p2pp/P2PP.py", line 91, in main
args['silent']
File "/home/mgorman/sites/p2pp/p2pp/mcf.py", line 827, in generate
convert_to_absolute()
File "/home/mgorman/sites/p2pp/p2pp/mcf.py", line 72, in convert_to_absolute
to_e = float(fields[j][1:])
ValueError: could not convert string to float: 
```

If wipe tower is disabled it works, but my slices don't account for any purging so they're all super short and they don't even get past the feed tube length (Sample 4 color keychain).

mjgorman, 2019-12-09:
Here's my Printer config: 
```
$ cat ~/.PrusaSlicer/printer/RailCore\ 300ZL\ P2SP.ini 
# generated by PrusaSlicer 2.1.0+linux64 on 2019-12-09 at 23:09:18 UTC
bed_custom_model = 
bed_custom_texture = 
bed_shape = 0x0,295x0,295x295,0x295
before_layer_gcode = ;BEFORE_LAYER_CHANGE\n;[layer_z]\n\n
between_objects_gcode = 
cooling_tube_length = 0
cooling_tube_retraction = 0
default_filament_profile = ""
default_print_profile = 
deretract_speed = 0,0,0,0
end_gcode = M106 S0 ; turn off cooling fan\nM104 S0 ; turn off extruder\nM140 S0 ; turn off bed\nG91 ; Relative move\nG1 E-1.0 Z+10.0 F1000.0 ; E : Nozzle Up/Bed Down 10mm and retract - (E to exclude from layer count)\nG90 ; Absolute moves\nG1 X1 Y290 ; Move Y-carriage for part removal\nM84 ; disable motors
extra_loading_move = 0
extruder_colour = #1A1313;#A80000;#00D423;#0B00EB
extruder_offset = 0x0,0x0,0x0,0x0
gcode_flavor = reprap
high_current_on_filament_swap = 0
host_type = octoprint
inherits = 
layer_gcode = ;AFTER_LAYER_CHANGE\n;[layer_z]\n;LAYER [layer_num]
machine_max_acceleration_e = 5000,5000
machine_max_acceleration_extruding = 1500,1250
machine_max_acceleration_retracting = 1500,1250
machine_max_acceleration_x = 5000,1000
machine_max_acceleration_y = 5000,1000
machine_max_acceleration_z = 500,200
machine_max_feedrate_e = 3600,120
machine_max_feedrate_x = 24000,200
machine_max_feedrate_y = 24000,200
machine_max_feedrate_z = 900,12
machine_max_jerk_e = 30,2.5
machine_max_jerk_x = 1000,10
machine_max_jerk_y = 1000,10
machine_max_jerk_z = 20,0.4
machine_min_extruding_rate = 0,0
machine_min_travel_rate = 0,0
max_layer_height = 0.2,0.2,0.2,0.2
max_print_height = 330
min_layer_height = 0.02,0.02,0.02,0.02
nozzle_diameter = 0.3,0.3,0.3,0.3
parking_pos_retraction = 0
print_host = 192.168.86.37
printer_model = 
printer_notes = Don't remove the following keywords! These keywords are used in the "compatible printer" condition of the print and filament profiles to link the particular print and filament profiles to this printer profile.\n\nPRINTER_VENDOR_RAILCORE\n\nPRINTER_MODEL_300ZL\n
printer_settings_id = RailCore 300ZL
printer_technology = FFF
printer_variant = 
printer_vendor = 
printhost_apikey = CEA1460978C142718A9BA4A22BE5CCE1
printhost_cafile = 
remaining_times = 0
retract_before_travel = 1,1,1,1
retract_before_wipe = 0.4%,0.4%,0.4%,0.4%
retract_layer_change = 0,0,0,0
retract_length = 0.4,0.4,0.4,0.4
retract_length_toolchange = 10,10,10,10
retract_lift = 0.18,0.18,0.18,0.18
retract_lift_above = 0,0,0,0
retract_lift_below = 0,0,0,0
retract_restart_extra = 0,0,0,0
retract_restart_extra_toolchange = 0,0,0,0
retract_speed = 65,65,65,65
serial_port = 
serial_speed = 250000
silent_mode = 0
single_extruder_multi_material = 1
start_gcode = ;P2PP PRINTERPROFILE=25f5bc11e0bad760\n;P2PP SPLICEOFFSET=30\n;P2PP MINSTARTSPLICE=130\n;P2PP MINSPLICE=80\n;P2PP MATERIAL_DEFAULT_0_0_0\n;P2PP MATERIAL_PVA_PVA_0_0_0\n;P2PP MATERIAL_PVA_PLA_0_0_0\n;P2PP MATERIAL_PLA_PLA_1_0_0\n;P2PP MATERIAL_PET_PET_0_0_0\n\n; even is you are not using SIDE TRANSITIONS, make sure the bed is defined correctly!!\n; the default settings will match Pruse MK3(s) printers.\n;P2PP BEDORIGINX=0\n;P2PP BEDORIGINY=0\n;P2PP BEDSIZEX=295\n;P2PP BEDSIZEY=295\n\n;P2PP PURGETOWERDELTA=0\n\n;Optional - waits for user to close window after processing\n;P2PP CONSOLEWAIT\n    \n; Following settings are optional (see description below)\n\n;P2PP LINEARPINGLENGTH=350\n;P2PP EXTRAENDFILAMENT=150\n\n; Following optional settings control the SIDE TRANSITIONING (see description below)\n\n;P2PP SIDEWIPELOC=X-45\n;P2PP SIDEWIPEMINY=120\n;P2PP SIDEWIPEMAXY=120\n;P2PP SIDEWIPECORRECTION=1.0\n;P2PP BEFORESIDEWIPEGCODE ;--ENTER GCODE TO BE EXECUTED BEFORE SIDEWIPE (one coomand per line,can be multiple lines)\n;P2PP AFTERSIDEWIPEGCODE ;--- ENTER GCODE TO BE EXECUTED AFTER SIDEWIPE (one coomand per line,can be multiple lines)\n;G1 Y145 F5000\n;G1 Y120 F5000\n;G1 Y145 F5000\n;G1 Y120 F5000\n;G1 Y145 F5000\n;G1 Y120 F5000\n;G1 Y145 F5000\n;G1 Y120 F5000\n;G1 Y145 F5000\n;G1 Y120 F5000\n;G1 Y145 F5000\n;G1 Y120 F5000\n\n; Multi-temp start code - 3 July 2019\n; Raise bed temp to profile requested bed temp\nM140 S[first_layer_bed_temperature] ; Raise bed temp to profile requested bed temp\nM104 S150 T0 ; Raise hot-end temp to 150C for G32 auto-calibration\nM190 S[first_layer_bed_temperature] ; Wait for bed temp to rise\nM109 S150 T0 ; Wait for hot-end temp to rise to 150C\nG28 ; home all axes\nG32 ; Autocalibrate bed\nG32 ; Autocalibrate bed a second time\nG32 ; And a third time ( for 4 point probe)\nG1 S2 Z20 F600 ; Move bed down 20\nM104 S[first_layer_temperature] T0 ; Raise hot-end temp to profile requested hot-end layer 1 temp\nM109 S[first_layer_temperature] T0 ; Wait for hot-end temp to rise to profile requested hot-end layer 1 temp\n;\nG1 X0.0 Y0.0 Z1.0 F1000.0 ; prepare to prime\nG92 E0 ; reset extrusion distance\nG1 Y120; Move to center of purge bucket\nG1 X-50; Move into purge bucket\nG1 E15 F200; Purge 15mm\nG1 Y145 F5000; wipe back and forth\nG1 Y120 F5000;\nG1 Y145 F5000; wipe back and forth\nG1 Y120 F5000;\nG1 Y145 F5000; wipe back and forth\nG1 Y120 F5000;\nG1 Y145 F5000; wipe back and forth\nG1 Y120 F5000;\nG1 Y145 F5000; wipe back and forth\nG1 Y120 F5000;\nG1 Y145 F5000; wipe back and forth\nG1 Y120 F5000;\nG1 Y145 F5000; wipe back and forth\nG1 Y120 F5000;\nG1 Y145 F5000; wipe back and forth\nG1 Y120 F5000;\nG1 Y145 F5000; wipe back and forth\nG1 Y120 F5000;\nG1 Y145 F5000; wipe back and forth\nG1 Y120  [... cut ...]
support_material_spacing = 2.5
support_material_speed = 125
support_material_synchronize_layers = 0
support_material_threshold = 0
support_material_with_sheath = 0
support_material_xy_spacing = 60%
thin_walls = 0
threads = 4
top_fill_pattern = rectilinear
top_infill_extrusion_width = 0
top_solid_infill_speed = 70%
top_solid_layers = 3
travel_speed = 150
wipe_tower = 1
wipe_tower_bridging = 10
wipe_tower_rotation_angle = 0
wipe_tower_width = 40
wipe_tower_x = -45
wipe_tower_y = 120
xy_size_compensation = 0
```

mjgorman, 2019-12-09:
Okay I believe I figured it out by re-reading things a bit and looking at the code. My origin as 0,0 but i had code for my start g-code that does a purge/wipe in my purge bucket at -45, 140 which seems to have been removed during processing which caused the issue. 

Should my origin really say -50,0 to fix this? Will that affect anything else?

tomvandeneede, 2019-12-10:
looks like the process has some issues with the input file when converting to absolute
could you please send me the unprocessed file so I can take a look?
If possible send though facebook PM

thanks

mjgorman, 2019-12-10:
Sent

tomvandeneede, 2019-12-10:
looking into it

mjgorman, 2019-12-10:
Fixed pushed to dev have resolved my issues. Thank you @tomvandeneede

closed 2019-12-10 by mjgorman