tomvandeneede/p2pp #48: Calibration question

GarthSnyder, 2019-12-09:
Just as a point of information, in the instruction "Prior to using the script it is imperative that you setup the printer according to the specifications set by Mosaic( this includes full calibration using canvas or Chroma)", what does "full calibration" mean, exactly?

I've completed the setup through the keychain calibration print (from CANVAS). But my understanding is that the calibration converges over time, so it seems to be commonly recommended to repeat the keychain print multiple times to allow the calibration to settle. Can this happen under p2pp, or does "full calibration" mean "your setup must already be fully converged before you start using p2pp"?

Or does "full calibration" mean something other than printing the keychain? Running the on-device calibration step as well? (The one that Mosaic tells you not to run unless you change hardware.)

tomvandeneede, 2019-12-10:
Fully calibrated means
1/ extruder is calibarted, this means steps, filament width, the whole set of things to check that what you think you extrude you are actually extruding
Over extrusion might be a problem when causing pressure to build up in the hot end and less material being deposited 
Under extrusion is not so much of a problem (other than making your print look bad)
Key thin is that if the extruder is asked to extrude 105mm of materia it does exactly that…no more no less..

2/ Palette needs to be calibrated…. although in the meanwhile I thing P2PP has matured to a point where you can use P2PP generated files to calibrate, It is not bad to use cnvas or chroma to do so… Make sure to put the extrusion rate to 100% when doing so.   if not you will end up with non-100 ping rates which are still consistent though (so no real harm)
Purpose here is to tedermine the loading length of your system for the initialisation

GarthSnyder, 2019-12-11:
Thanks - I will prep an abbreviated version of this as a PR.

closed 2019-12-19 by GarthSnyder