tomvandeneede/p2pp #50: Fails to increase Z back to print height when using PURGETOWERDELTA

mjgorman, 2019-12-14:
After the first purge to the tower (after the initial layer) the print several layers in the printer fails to go back up to the Z height it was at before lowering to the tower for purge, causing it to move back and bash into the model (or supports in my case) and start printing again at this lower height. 

Unprocessed Gcode:
https://drive.google.com/file/d/11UOdeYK5N3cAy_aR6CEcezzTv7hB-Jug/view?usp=sharing

Processed Gcode:
https://drive.google.com/file/d/1FgJZJjdj_vUA0tg-EbYNF3FCX1NZStlu/view?usp=sharing

tomvandeneede, 2019-12-15:
Michal, 

Could you specify at which layer this happens approx.
I see all the tower moves prefixed and suffixed with Z- moves as should

tomvandeneede, 2019-12-20:
added check for required Z-hop on retracts

closed 2019-12-20 by tomvandeneede