tomvandeneede/p2pp #52: Color where it doesnt belong

audiorazor, 2019-12-20:
A few people, including myself, have been having issues where a strip of color ends up where it doesnt belong.   I never used to have a problem until recently and am trying to figure out if p2pp is causing the issue or its something else.   I have emailed Mosaic to see if its a problem with my machine and figured I would post here also.
Attached is the gcode.  Not sure if that can tell you anything.  

[Mini John McClane fixed purge.mcf.gcode.zip](https://github.com/tomvandeneede/p2pp/files/3986344/Mini.John.McClane.fixed.purge.mcf.gcode.zip)

tomvandeneede, 2019-12-20:
Hi,

more interesting here would be the pings of your print.   It appears your extruder might be a little off and P2 tends to correct for that during the print (this correction is reflected in pings going up and down).   If at some point the correction reaches a certain threshold, the color will be off.

I can only remain vague over this as the actual algorithm to perform these corrections is not disclosed by Mosaic… hence there is very little P can do in P2PP to prevent this from happening.

What is important though is that you perform proper extruder calibration, your first layer is well calibrated and you don’t touch the flow rate controls on your printer or in octoprint during the print.  

I am happy to look into your ping results and octoprint log file for a print that has missed lines to see if there is anything to spot.

audiorazor, 2019-12-20:
[octoprint.log](https://github.com/tomvandeneede/p2pp/files/3986727/octoprint.log)
[Mini_John_McClane_fixed_purge.txt](https://github.com/tomvandeneede/p2pp/files/3986729/Mini_John_McClane_fixed_purge.txt)
I have attached the octoprint log and ping files.   Thank you for your time.  I'm fairly sure its not p2pp thats causing my issue but I am just trying to track it down.   

I have a second Palette2 that doest appear to be having any issues.   I'm going to try printing on that machine and see if I get the same results.

tomvandeneede, 2019-12-20:
Then likely the calibration of one might be ever so slightly off.
I will look a the log files and see if I can find something

at first sight your pings look very stable, no big jumps
Are you printing with high Z on first layer? could explain the slightly raised first, but this should not be an issue

model looks quite palette friendly as well… some shorter splices but not a whole bunch in one after the other

could you send a view of your pongs… from the octoprint log it appears they are 99% which is weird as mostly all printers/printer combinations I have seen report pongs 100% +/- 0.05, but that could be coincidence… you can also check what they look like on your other P2

audiorazor, 2019-12-20:
My second palette's pongs are 99% also.  I have attached a picture.

I think the high ping in the beginning is because of how I load the
filament at the start.  Requires me to pull the filament out of the ptfe
tube a bit.

I will try this print on my other machine to compare and also hope that
Mosaic gets back to me soon.

tomvandeneede, 2019-12-20:
Will close this issue as not reallt p2pp related.   If the issue continues, don't hesitate to join the Mosaic Fanufacturing Facebook group ... there is a ton of knowledge there.  We also have a P2PP specific group should you need any P2PP specific help

closed 2019-12-20 by tomvandeneede