tomvandeneede/p2pp #53: Blobs in print

EldarAron, 2019-12-28:
Hello
I was recommended by Mosaic to try out P2PP because a print I want to make to make times out when slicing one Canvas. I really like the potential of this script!

I'm trying to get P2PP to work. I made the multi colour headphones holder through Canvas3d.io and it worked great. Then I tried to make the same model with PrusaSlicer and P2PP.

But for days now I have been struggling with blobs. Prusa makes a tuning line along the front edge. When it has done that it stops and makes a blob. Then it starts on the wipe tower. That's ok. Then it starts on he model. After finishing the first layer it makes another blob on the model. The first time I tried to print the model, the printer hit the blob and then went into crash mode and recovered. 

I have tried multiple things. Remade profiles (printer, filament and print). Nothing. Looked at the P2PP settings. In the video he says to deactivate "Ramming parameters". That is not mentioned anywhere in the instructions. I have tried with and without the ramming setting. No help.

What am I doing wrong? 

Keep up the good work and happy holidays! 🎄🎆☃️

//EldARon

[image]
[earbub_body_0.2mm_PLA_MK3S_2h9m_v3.mcf.gcode.txt](https://github.com/tomvandeneede/p2pp/files/4007549/earbub_body_0.2mm_PLA_MK3S_2h9m_v3.mcf.gcode.txt)
[printer_config.ini.txt](https://github.com/tomvandeneede/p2pp/files/4007551/printer_config.ini.txt)

tomvandeneede, 2019-12-29:
Hi ,

could you please send me the good before processing. 
there appears to be something extra in the print before the layer starts but it would be a lot easier to start from the original gcode to confirm the issue is not in there yet

I also see you are still using 3.6.8… there are some known issue with that version so I suggest you move to 3.6.17 ot 4.0 development version and start from there

Regards

EldarAron, 2019-12-29:
Hey

Thanks for you quick response. I thought I had the latest version, but I didn't, as you mentioned.

Upgraded 3.6.17 and tried again. This time no blob. So it's ok now! So much less waste with the "PURGETOWERDELTA" setting! 

But what is the correct settings for "Ramming Parameters". You don't mention it in det description but Kurt mentions it in the video. 

Thanks again.

//EldARon

tomvandeneede, 2019-12-29:
this is what I use.

Regards
Tom

EldarAron, 2019-12-29:
Ok, thanks! Keep up the good work! 😄

🥂🎆

//EldARon

closed 2019-12-29 by EldarAron

tomvandeneede, 2019-12-29:
Thank you very much!

gdombiak, 2019-12-29:
Was just watching the video and noticed the "Ramming Parameters" and this fresh new issue was posted. :) How should we configure "Ramming Parameters"  in PrusaSlicer?

Thanks,
Gaston

tomvandeneede, 2019-12-29:
Put all to 0

Ramming is only there to get the point to the tip and to get filament properly in and out

We don’t need all of this

P2PP filters all of this out but the less there is in the less confusion it can cause