tomvandeneede/p2pp #54: [Marlin 2.0.1] Extruder not moving in purge tower

gdombiak, 2020-01-04:
Hooked up PrusaSlicer 2.1.1 and P2PP commit 20ad5ff470bb4d2e87bea25da5cc19726408c1aa with my MK3 and works just fine. I then switched printers to use it with my Ender 3 that is running Marlin 2.0.1 and noticed that the purge tower that should be printed after object was printed does not move extruder. Once nozzle moves back to the object it resumes extruding. Goes back to purge tower and it moves in X/Y but extruder does not move. and so on until I cancel the print job.

Been looking at the generated gcode and cannot find a reason for not extruding in purge tower AFTER object was printed in the same layer. It prints whatever part of the purge tower that goes BEFORE the object. I see that feedrate is higher than the one I have configured in PrusaSlicer but the same happens with the MK3 and it prints just fine. Looking at Marlin's source code to see if that could be the reason but so far no luck. I did not see any error printed to the terminal (even though it moved super fast) while trying to extruder on the purge after.

Tried a few different models and same thing. I initially thought nozzle was too close to bed when I saw this on the first layer or bed not leveled but that is not the case. This happens on the second and third layer as well so not the case. 

I attached an example of the gcode I'm printing on the Ender 3 that has this problem. Printing from Canvas (in another printer profile) works fine.

[E_PLA base.mcf.gcode.zip](https://github.com/tomvandeneede/p2pp/files/4021628/E_PLA.base.mcf.gcode.zip)

Thanks,
Gaston

tomvandeneede, 2020-01-04:
> Hooked up PrusaSlicer 2.1.1 and P2PP commit 20ad5ff <https://github.com/tomvandeneede/p2pp/commit/20ad5ff470bb4d2e87bea25da5cc19726408c1aa> with my MK3 and works just fine. I then switched printers to use it with my Ender 3 that is running Marlin 2.0.1 and noticed that the purge tower that should be printed after object was printed does not move extruder. Once nozzle moves back to the object it resumes extruding. Goes back to purge tower and it moves in X/Y but extruder does not move. and so on until I cancel the print job.

Gaston,

If the federate in the purge tower is an option, try using the purge tower speed settings (just in test added few days ago, requires latest version)

;P2PP PURGETOPSPEED=2000

this will cap the topspeed at 2000nn/min or approx 33mm/sec
you can go lower.

see if this fixes the problem.   I have not had any similar reports before.

tomvandeneede, 2020-01-04:
also check if the feed tube does not get obstructed by moving to the purge tower location

I looked at the print and it should be extruding… if anything it is indeed the speed …. but you should get at least some extrusion…. getting nothing is plain weird  Also the bottom layer is printed fast er but as a lot lower rate than upper layers so.   If with MK3S the same model works fine, It is likely an issue with profile still for your ender printer.

gdombiak, 2020-01-04:
Thanks Tom for the blazing fast response. I will play with that setting and let you know. 

Here is the latest progress. I copied the releavent gcode to a new file so I can print single color and without Palette. I can now test way faster and hopefully isolate the problem. The good news is that I was able to reproduce with this reduced gcode: 
[test_tower.gcode.zip](https://github.com/tomvandeneede/p2pp/files/4021669/test_tower.gcode.zip)

If I remove what is between **START OBJECT** and **END OBJECT** then the tower gets correctly printed. If I leave the gcode to print the object then there are no more extrusion movements on the tower AFTER the object was printed. Will keep isolating the problem and report back.

Let me know if you see something before my slow analysis process.

Thanks,
Gaston

gdombiak, 2020-01-04:
Closing the ticket since this has nothing to do with P2PP. I'm still researching what triggers the problem but I see the problem now. I'm using OctoPrint and have the CancelObject plugin installed. I noticed that certain movements (under what circumstances is still under research) were NOT sent from OctoPrint to the printer. I looked at OctoPrint's logs and indeed I found lots of entries like this:

> 2020-01-04 17:20:17,345 - octoprint.util.comm - ERROR - Error while processing hook cancelobject for phase queuing and command G1 Y140.344 E1.85090:
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python3.8/site-packages/octoprint/util/comm.py", line 3333, in _process_command_phase
    hook_results = hook(self, phase, command, command_type, gcode, subcode=subcode, tags=tags)
  File "/home/pi/oprint/local/lib/python3.8/site-packages/octoprint_cancelobject/__init__.py", line 480, in check_queue
    if e_move[0] > obj["max_x"]:
TypeError: '>' not supported between instances of 'NoneType' and 'float'

This is a bug in that plugin. Will keep analyzing and hopefully creating a PR for that plugin.

Thanks Tom for your time and this great P2PP project.
Gaston

closed 2020-01-04 by gdombiak

tomvandeneede, 2020-01-04:
did you get it solved?

gdombiak, 2020-01-04:
yep, have a PR for the CancelObject plugin. I guess it was not tested to be used with purge towers. It always assumes that G1 moves include X and Y coordinates. So a move like this makes it crash and the exception is not trapped so OctoPrint does not send the gcode to the printer.

> G1 Y140.344 E1.85090

tomvandeneede, 2020-01-04:
thanks for confirming and testing.

good luck getting the issue solved.   CancelObject is a neat featured

gdombiak, 2020-01-04:
yep, super cool plugin. That is why I added support for it in OctoPod (the iOS app that I developed).

tomvandeneede, 2020-01-04:
Wow you develop IOS apps.. I envy you
what platform do you use to develop? Xcode? HTML5?
I really would like to start doing st like that as well

gdombiak, 2020-01-04:
no need to envy. Your P2PP is rock solid and super useful. I'm not really an iOS developer (or a mobile dev or a front end dev. Just a backend guy) but as my hobby I taught myself Swift which is straight forward. I use MacOS to develop in Swift. XCode is the IDE that is free. If you have a Mac then you can run OctoPod in your local environment and run the app in your iPhone/iPad simulator. I think that Apple now lets you deploy it to your phone as well. You will need to pay the $99 per year if you want to use services that rely on Apple's services (e.g. push notifications) which OctoPod uses.

Happy to help set you up in your env if you want to dig into it. I think that OctoPod could be a great intro to this world as well since you can add small features to a running app (that is open source)

tomvandeneede, 2020-01-04:
I used to pay 99 per year to apple but I found with Xcode 8 the changes were coming in to steep.  
thanks for ths info :-)

gdombiak, 2020-01-04:
Sure thing. Let me know if you want to become an OctoPod dev. Happy to help! :D