tomvandeneede/p2pp #56: Side purging every few layers when there are no tool changes.

mmontminy, 2020-01-26:
I was testing 4.0.8 with PS 2.2a2 last night, and noticed that the print was doing a small purge every few layers even when there was no tool change.

The print had no tool changes until 3/4 of the way through, yet a purge happened at layer 8'ish, and then every dozen or so layers after.

It's entirely possible I had bad settings, as I noticed from the docs that there are several new settings. I incorporated those I thought made sense or were needed.

I don't know if it's related to bad settings, a bug, or just a bad print, but the color changes were also shifted when they did start.

I went back to 2.1 and 3.0.6, but wanted to share the gcode in case there are bugs to be found/squashed. If it's operator error, feel free to close this out.
Attached is the gcode.
[Dog.mcf.gcode.zip](https://github.com/tomvandeneede/p2pp/files/4113510/Dog.mcf.gcode.zip)

tomvandeneede, 2020-01-26:
can you please provide the unprocessed file so I have a file with all correct triggers

mmontminy, 2020-01-26:
Unfortunately I didn't keep it :( I could attempt to recreate it, but I don't know that I would do so correctly. I've since reverted back to 3.6.0RC and PS2.1 and re-saved my 3mf with those settings.

mmontminy, 2020-01-26:
I'm going to setup my other PC with PS2.2.0a2 and P2PP latest master and re-do the print.

FYI, there looks like a gap between the 3.6.0RC in the releases section and what the latest stable of the 3.6 line was. I think I was running 3.6.17 which you had fixed some side purging issues in. I don't see that release, and now 4.0 has taken over master and dev.

Maybe 3.6 was the most stable release for PS 2.1?

tomvandeneede, 2020-01-26:
Mark,

Added an extra check to 4.0.8 to avoid extra purge
I have no project to confirm the test so it would be great if you could send me the gcode file without P2PP processing so I can confirm the behaviour

mmontminy, 2020-01-26:
I had to re-install and re-slice, so I can't confirm the settings are the same. If I had to guess, I'd suspect something related to the AUTOADDPURGE setting. It's not clear if that's good to use or not, and I believe it's in use for this print, as there's a number of short splices that are all magically purging exactly 80mm.

I kept the pre-processed gcode from temp this time, so I have both. I just started the print... but if you'd like to look to see if there's anything off in advance, here they are: 
[Gcode.zip](https://github.com/tomvandeneede/p2pp/files/4113662/Gcode.zip)

mmontminy, 2020-01-26:
Looks like 4.0.8 didn't fix it, it just did a purge with no tool change needed.

tomvandeneede, 2020-01-26:
please give me the unpriced gcode for this print then… should make an easy fix

mmontminy, 2020-01-26:
It's in the attached Gcode.zip 2 comments back. Unless I'm misunderstanding which gcode you want. That zip has the code PS saved before P2PP processed it, as well as a copy after processing. I can also attach the 3mf if you want that.

mmontminy, 2020-01-26:
Either operator error, or there's something weird happening when adding LAYERHEIGHTZ to the after layer change gcode.

If that's there, no matter what I set purge amounts to, I get short splice warnings. If I remove that, the short splice warnings go away.

According to the PDF, that's now required as of 4.0.4, but adding it seems to cause problems.

tomvandeneede, 2020-01-26:
pls send me the goode unprocessed

can you please check 4.0.8.5 it removes the unnecessary purges
Looked at the output and seems better now.

tomvandeneede, 2020-01-26:
did you add it exactly as it was specified?

mmontminy, 2020-01-26:
I copied and pasted out of the PDF. If that line is there I get odd short splice warnings.

When I reverted back to 3.6.0RC last night I couldn't figure out why I couldn't slice anything without short splice warnings on layer 1, which had no spliced at all. I discovered it was due to that line eventually, and assumed it was confusing 3.6.0 and removed it.

But I'm seeing the same behavior with 4.0.8.5, if that's there, I get short splice warnings on many layers and no amount of purge length resolves it.

I removed it to see if that was affecting the early purging, but had to abort the print since my cooling shroud mount broke /sigh.

tomvandeneede, 2020-01-26:
any chance getting the gcode unprocessed version.
There are really too many settings combinations to start from a blank template

mmontminy, 2020-01-26:
It's been there since this morning, in the Gcode.zip file. But here it is again in case something happened to that zip.

This was after slicing, before P2PP processing. I didn't get short splice warnings with this, because it had the AUTOADDPURGE setting.

My theory is that LAYERHEIGHTZ is messing with the splicing somehow, creating short filament lengths, and AUTOADDPURGE is masking it by setting the splices to 80mm. 

But I'm just guessing, you certainly know how this works better than I.

[Dog-pre.zip](https://github.com/tomvandeneede/p2pp/files/4114102/Dog-pre.zip)

mmontminy, 2020-01-26:
Got my printer back together, running 4.0.8.5, with AUTOADDPURGE and LAYERHEIGHTZ removed from the start and after layer g-code respectively...

I'm 30 layers in without any unnecessary purging happening. I don't know which of the 3 changes addressed the issue, so I'll re-try again tomorrow with the 2 settings back in the gcode. Sooner if the print fails.

mmontminy, 2020-02-01:
I finally have some time to get back to this. The original bug report appears to be the result of another issue. There seems to be something happening with the LAYERHEIGHT settings is placed in the after layer gcode per the docs.

If I have: ;LAYERHEIGHT [layer_z] as the last line of my after layer change gcode, P2PP throws short splice errors starting at layer 36 even though the first splice doesn't happen until around layer 256. If I remove LAYHERHEIGHT, the short splice errors go away, and the resulting gcode seems fine. It printed, changed colors as expected. It ultimately failed due to an outlying ping variance.

When you also have AUTOADDPURGE, this masks those false short splice errors by making them 80mm, and I suspect because it thinks there's a splice at that point, it caused the original problem of purging when none was expected.

Here's the gcode output without being processed by P2PP: 
[Dog-pre.zip](https://github.com/tomvandeneede/p2pp/files/4142755/Dog-pre.zip)

And here's the same gcode processed by P2PP: 
[Dog.mcf.gcode.zip](https://github.com/tomvandeneede/p2pp/files/4142757/Dog.mcf.gcode.zip)

Lastly, if it helps debug, here's the 3MF for the project showing how the profiles are setup, in case I've made a mistake in configuration that causes this behavior: 
[Dog.zip](https://github.com/tomvandeneede/p2pp/files/4142758/Dog.zip)

This is possibly related to this Facebook post, the symptoms are similar: https://www.facebook.com/groups/591271211347189/permalink/814216585719316/

tomvandeneede, 2020-02-01:
Thanks for supplying the files.  There was indeed a line that was in the wrong place.
Hope this is what it needs to be
No more errors of short splices.

mmontminy, 2020-02-01:
Is there a newer version I should try? The master branch hasn't been updated, was the fix pushed to dev, or you've not pushed a fix yet?

tomvandeneede, 2020-02-01:
Made the changes in the dev branch

mmontminy, 2020-02-01:
That appears to have resolved it.

closed 2020-06-23 by tomvandeneede