tomvandeneede/p2pp #57: Side purge speed way too fast

jessegrunert, 2020-02-02:
I just recently got my printer set up for side transitions. When I set it up, everything worked fine, the head moved to the right position, however the feed rate was crazy high. Basically causing my extruder to instantly skip and very little filament was actually extruded. I had set a narrow y-min/y-max (200/300) as I didn't want the y axis to move around too much.

Is there a value or setting that I'm missing to control this?

tomvandeneede, 2020-02-02:
there are 2 commands
WIPEFEEDRATE=xxxx sets the wipe federate in mm/min (movement)

> On 2 Feb 2020, at 09:06, Jesse Grunert - Acro Media <notifications@github.com> wrote:
> 
> e a value or setting that I'm missing to control this?

jessegrunert, 2020-02-02:
Thanks, I didn't realize this would relate to the sidewipe as well. Confirmed that it works as expected. Thanks again for writing such an awesome piece of code!

closed 2020-02-02 by jessegrunert