tomvandeneede/p2pp #58: Failing to process due to g.E not being set

martinjaroschek, 2020-03-01:
Just ran into this one while processing gcode from PS 2.1.1:

Traceback (most recent call last):
  File "C:\p2pp-master\\p2pp.py", line 145, in <module>
    main(vars(arguments.parse_args()))
  File "C:\p2pp-master\\p2pp.py", line 91, in main
    args['silent']
  File "C:\p2pp-master\p2pp\mcf.py", line 876, in generate
    gcode_parseline(process_line_count)
  File "C:\p2pp-master\p2pp\mcf.py", line 525, in gcode_parseline
    if not (coordinate_in_tower(_x, _y) and coordinate_in_tower(v.purge_keep_x, v.purge_keep_y)) and g.E > 0:
TypeError: '>' not supported between instances of 'NoneType' and 'int'
Press any key to continue . . .

Not sure where g.E is supposed to get its value, but a simple sanity check in line 525 works around this ("and g.E" as additional condition)

tomvandeneede, 2020-03-01:
hi

can you please send me the files that generate this errors so I can have a look
the error alone is rather little to work form

martinjaroschek, 2020-03-01:
hi,

Attached the PrusaSlic3r project:

[Body7_.zip](https://github.com/tomvandeneede/p2pp/files/4272404/Body7_.zip)

tomvandeneede, 2020-03-02:
Hi,

In order to work I would really need the gcode file, there are too many things  that mak the file go different when replacing the project 

regards

tomvandeneede, 2020-06-23:
missing info

closed 2020-06-23 by tomvandeneede