tomvandeneede/p2pp #59: palette+ in accessory mode fails to export. IOError: [Errno 13] Permission denied: 'tmpfile

robtrama, 2020-03-31:
i followed the setup for palette+
using prusaslicer 2.2 
python 2.7.17

Traceback (most recent call last):
  File "C:\p2pp-master\\p2pp.py", line 145, in <module>
    main(vars(arguments.parse_args()))
  File "C:\p2pp-master\\p2pp.py", line 91, in main
    args['silent']
  File "C:\p2pp-master\p2pp\mcf.py", line 921, in generate
    with io.open('tmpfile', 'w', newline='\r\n') as opf:
IOError: [Errno 13] Permission denied: 'tmpfile'
Press any key to continue . . .

tomvandeneede, 2020-03-31:
hmm 

this looks like a config issue on your PC.  What is the filename of the file?  Maybe try saving to a location with no spaces etc should that be the case

> On 31 Mar 2020, at 14:16, robtrama <notifications@github.com> wrote:
> 
> t

robtrama, 2020-03-31:
Currently the file name is just “test” and I can try saving it directly to the desktop and see what that does

Thanks for the quick reply

Robert Trama
Fodera Guitars Partners
Rtrama@Fodera.com
646-734-3655

tomvandeneede, 2020-03-31:
Hi Not just the file name, also the folder name

are you running on windows or macOS?

tomvandeneede, 2020-03-31:
I may have found something, can you please download the latest dev version and give it a spin?

robtrama, 2020-03-31:
I’m running windows 10 and yea it’s in a few subfolders that have space in the name

Robert Trama
Fodera Guitars Partners
Rtrama@Fodera.com
646-734-3655

tomvandeneede, 2020-03-31:
Just give the new version a try and see where this gets you

robtrama, 2020-03-31:
I can do that in a couple hours. The version I’m running I downloaded last night.  

Thanks again

Robert Trama
Fodera Guitars Partners
Rtrama@Fodera.com
646-734-3655

tomvandeneede, 2020-03-31:
you only have to replace the file p2pp/mcf.py if you have the latest dev version already

robtrama, 2020-03-31:
Got it

Robert Trama
Fodera Guitars Partners
Rtrama@Fodera.com
646-734-3655

robtrama, 2020-03-31:
Still the same error

Robert Trama
Fodera Guitars Partners
Rtrama@Fodera.com
646-734-3655

tomvandeneede, 2020-03-31:
cn you please send me the unprocessed gcode file if possible, I will try it here then

tomvandeneede, 2020-03-31:
right before generating the file it should say Generating PALETTE MAF/MSF file:
followed by a filename.   Would you mind copying that full line?

thanks

robtrama, 2020-03-31:
i tried but it doesnt keep the p2pp app on long enough for me to copy it
it closes out almost immediatly

robtrama, 2020-03-31:
managed to record it and get the output file
Generating PALETTE MAF/MSF file: c:\users\admin\Desktop\body.mcf.maf

tomvandeneede, 2020-04-01:
Any chance you can temporarely disable the script (in output optioned, remove the script name) and send me the gcode that is generated by prusa slicer
This will allow me to run the pcode here and see why it fails

robtrama, 2020-04-01:
Is did that and sent it over last night. I had to zip the file. 

Robert Trama
Fodera Guitars Partners
Rtrama@Fodera.com
646-734-3655

tomvandeneede, 2020-04-01:
Hi Robert,

can you PM it in Facebook? maybe the easiest way.   Did not come trough in the email

robtrama, 2020-04-01:
https://drive.google.com/open?id=11ANpnn8KZq-XBpccjLz8nYN1BPR2Zdel

messenger was giving me a hard time  here is a link to it

tomvandeneede, 2020-04-01:
I am not getting an error here but I am working on a mac so may be the reason
this is the output I am getting
Will see if I can setup a windows machine tonight or tomorrow and these there

  Python Version Information: 3.7.0
  Reading File /Users/tomvandeneede/Desktop/body.mcf.gcode
  Analyzing slicer parameters
  File was created with PS version:2.2.0
  Pre-parsing GCode
  Splice Offset set to 30.00mm
  Config: Palette2 Accessory Mode Selected
  Extra filament at end of print   150.00mm
  Creating tool usage information
  Generate processed GCode
  The PRINTERPROFILE identifier is missing, Please add:
;P2PP PRINTERPROFILE=<your printer profile ID>
to your Printers Start GCODE.
  No or Invalid Printer profile ID specified
using default P2PP printer profile ID 50325050494e464f
  Generating GCODE file: /Users/tomvandeneede/Desktop/testo.gcode
  Generating PALETTE MAF/MSF file: /Users/tomvandeneede/Desktop/testo.maf

tomvandeneede, 2020-04-01:
> I started to suspect that it’s something to do with the python install on my machine. I used both 2.7 and 3.8 with the same error
>
> Robert Trama
> Fodera Guitars Partners
> Rtrama@Fodera.com
> 646-734-3655

Is this a work PC or your personal one.   Try saving the file somewhere under my documents
I also saw that the filename was a little weird so will definitly continue to test further under windows

robtrama, 2020-04-01:
Personal pc,  the post processor works fine when it’s not in accessory mode. As if I was generating a file for a P2. It’s only in the accessory mode that it has an issue. I have an old Mac I will try and set it all up on   If it can handle it

I will try saving it under my docs

Robert Trama
Fodera Guitars Partners
Rtrama@Fodera.com
646-734-3655

tomvandeneede, 2020-04-01:
Yep it uses a different output module for maf files because of issues we had before.

robtrama, 2020-04-01:
It’s a shame chroma won’t accept files that already have a purge tower in
them and just generate the .seem file from that

nb1767, 2020-05-22:
Hello!
Has anyone come up with a solution to this issue? I'm having the exact same problem trying to use accessory mode with a Palette 2S Pro. I get the same error message as shown in the first post in this thread. As stated by robtrama, mine also works fine when I try to use connected mode, but doesn't work in accessory mode (which is how I use my Palette). Thank you!

tomvandeneede, 2020-05-22:
could you please send me your unprocessed gcode file so I can have a go at it

> On 22 May 2020, at 20:30, nb1767 <notifications@github.com> wrote:
> 
> sory mode with a Palette 2S Pro. I g

robtrama, 2020-05-22:
I’ve only gotten it to work using a Mac and even then the file it outputs is not compatible with the palette/+. 

Robert Trama
Fodera Guitars Partners
Rtrama@Fodera.com
646-734-3655

tomvandeneede, 2020-05-23:
OK No problem, I work on a MAC too so I should be able to simulate.  Any chance getting me the gcode you want to process so I can repeat the issue here.

tomvandeneede, 2020-05-23:
I redid the output based on the same library as the other files.
could you please retry the dev version

tomvandeneede, 2020-05-23:
redid the output of the maf file
could you please check the dev version (4.7.0)

tomvandeneede, 2020-06-23:
closed due to no feedback

closed 2020-06-23 by tomvandeneede