tomvandeneede/p2pp #60: Latest dev build crashes

audiorazor, 2020-05-13:
[image]

Not sure whats causing it.  Python opens and it looks like its about to begin and then I get that error.

tomvandeneede, 2020-05-13:
Hi,

can you please send me the geocode file you are trying to process

audiorazor, 2020-05-13:
[desert_-_1.mcf.pdf](https://github.com/tomvandeneede/p2pp/files/4621807/desert_-_1.mcf.pdf)

The pdf thats attached is from using the master version of p2pp that was able to process it.

tomvandeneede, 2020-05-13:
Hey

looks like an oder version

;Generated with P2PP version 4.00.008

can you please update to the latest (4.6.0).   This issue was actually solved in 4.5.0 and was related to the preview image encoded in the Prusa Mini file gcode file that contains a specific pattern that throws off the parsing.

Regards

audiorazor, 2020-05-13:
Where do I get that version?  I am downloading the master zip file and thats the version its giving me.   I'm on a mac btw.

tomvandeneede, 2020-05-13:
You should not download Master but where it says Branch in the top left of the window, there you must select DEV and then get zip files or installation files
I did upgrade it to 4.6.0 to make a certain version cutoff so it should be easily recognised when correct version is operational.

audiorazor, 2020-05-13:
Weird, I downloaded the dev version this morning and it crashed.  Now since talking to you it works fine.  Im assuming you did something to fix it!    
Would you be able to update the description on the page?  You used to update it with the date and what you changed and that was a good reference for me.  

Anyway, thank you.

tomvandeneede, 2020-05-13:
Nope, only changed version to 4.6.0 as I was unsure I did it before uploading

audiorazor, 2020-05-13:
Well, thank you even if you say you didnt do anything!  Keep up the good work.  Dont know what I would do without this bit of software!

closed 2020-05-13 by audiorazor