tomvandeneede/p2pp #61: MFC

Kelvinator78, 2020-06-23:
Although I add the .mfc.gcode for the file extensions and sliced it to my SD card to put in the Palette 2, it wouldn't see what I did as the file format is gcode and not mfc

tomvandeneede, 2020-06-23:
Hi,

Can you please be a little bit more specific on what you are trying to achieve.
*/ mcf feels are still gcode files, they are just extended with additional data…please PM me the file and I  will take a look
*/ can you confirm p2pp has worked (popup screen)

Kelvinator78, 2020-06-23:
Now it says WARNING: No Algorithm defined for transitioning ABS to ABS.
Using Default.

Also on the manual under Palette+ setup when I have a Palette 2 Pro it says
that it will generate a Gcode file as well as a MSF file and there is a
script code for it but don't know where to copy and paste it.

Kelvin Case

tomvandeneede, 2020-06-23:
ok so you are now going from ABS to ABS in your print.   If you do so you need to make sure to specify for the Palette how to splice ABS to ABS
 
you will need to add a line 

;P2PP MATERIAL_ABS_ABS_[Heating]_[Compression]_[reverse] 
to your print with the desired parameters.

Kelvinator78, 2020-06-23:
On the manual under Palette+ setup when I have a Palette 2 Pro it says that
it will generate a Gcode file as well as a MSF file and there is a script
code for it but don't know where to copy and paste it.

tomvandeneede, 2020-06-23:
All config goes into the startup gcode

Please use the facebook support group for your follow up questions as I am not able to always monitor this account

regards

tomvandeneede, 2020-06-23:
Looks like a setup issue, please use the facebook group to further discuss

closed 2020-06-23 by tomvandeneede

Kelvinator78, 2020-06-23:
I have figured it out. In the website manual it says you have to save it as
filename.mfc.gcode which I was doing. I now have saved it as filename.maf
and that worked

tomvandeneede, 2020-06-24:
You should not have to specify filenames.   The programme should create 2 files by itself one gcode for the printer and one msf/maf file for the P2 memory card
this is assuming that you want to print in accessory mode.   The problem specification has been rather vague.

Please use facebook for support.