tomvandeneede/p2pp #62: BB3D - Purge volumes are not taken into consideration

anyk91, 2020-07-01:
Hello

I have noticed that no matter how much purge volume i select between individual colors, it always generates 2 blobs.

I have experimented with 80/120/160 values in simple and advanced mode of PS2.2.0

tomvandeneede, 2020-07-01:
Can you please share the gcode file

anyk91, 2020-07-01:
attached settings and gcode file. also 3mf file if needed
[test.zip](https://github.com/tomvandeneede/p2pp/files/4858079/test.zip)

tomvandeneede, 2020-07-01:
Just looking at the file the amount of purge checks out with the amount generated in Prusa Slicer.
Can you check with the purge volume on the bed and see if the block actually changes with the amount of purge
Also other features like purge to infill or purge to object may throw the tower off

anyk91, 2020-07-01:
I think i found my answer. You are using mm if filament in script and PS is using mm^3. I set much higher values and those have changed to 3 blobs. It took a bit of playing with the volumes to get it to work right. So when I used 120mm^3 it only ended up being ~50-60mm and created 2 blobs not 3.
I haven't used any other purge to infill or object options.

I'll try the print now. 

Not sure how possible it is (or if it even makes sense) but it would be amazing if we could add P2PP config lines that would specify amount of blobs between filament numbers, i.e:

;P2PP BIGBRAIN3D_1_TO_2=2
;P2PP BIGBRAIN3D_1_TO_3=3
;P2PP BIGBRAIN3D_1_TO_4=3
;P2PP BIGBRAIN3D_2_TO_1=2
...

tomvandeneede, 2020-07-01:
would be a great idea but it would be of limited use when doing purge to infill etc 
I would still be able to calculate this directly from the matrix information.   it is embedded at the end of the gcode file
I tend to leave the work as much as possible to prusaslicer to retain maximal compatibility

anyk91, 2020-07-01:
I see, that makes sense, I think the use matrix feature would be amazing for anybody NOT using those purge into object/infill options.
Maybe to avoid config errors throw an error when any of those are enabled:

; wipe_into_infill = 1
; wipe_into_objects = 1

and switch like that is enabled

;P2PP BIGBRAIN3D_USE_VOLUME_MATRIX=1

I know it took me good 5-10 min to figure out best matrix combinations to ensure I get 3 blobs from black to white and 2 from white to yellow. 
At least I know what to look for now and make it do what I want

audiorazor, 2020-08-08:
@anyk91 Any chance you can post what purge settings you used to have success?  I am about to try to use BigBrain again and would love a little guidance.

anyk91, 2020-08-08:
> @anyk91 Any chance you can post what purge settings you used to have success? I am about to try to use BigBrain again and would love a little guidance.

Unfortunatelly I dont recall exact settings, I was basically guesstimating those and increasing to have enough blobs. You simply would want to look at each color change and see how many blobs it generates. Script is pretty verbose.

closed 2020-12-20 by tomvandeneede