tomvandeneede/p2pp #63: Material overrides not applying correctly

drsandytoes, 2020-07-23:
I was having a hard time getting my material overrides to be recognized. Only the last filament would ever show a non-default value. This appears to be because mcf.parse_gcode() calls parameters.check_config_parameters(), but v.current_tool has never been set; it's still -1. v.current_tool doesn't get changed until after the pre-process step is done. It looks like parse_gcode() does watch for tool changes, but it affects a local variable (cur_tool). So either this could be passed in, or we could assign v.current_tool in this method (and reset it back to -1 at the end for safety?). If you have a preference, I could put up a pull request.

Separately, I'm not sure that the output is actually using the right settings... What is the algorithm table supposed to look like? I see things like "D41 D0002 D0000 D0001" but does D41 indicate extruder 4 to extruder 1, or are 4 and 1 just arbitrary material types that are assigned to an extruder separately.

tomvandeneede, 2020-07-23:
Can you please send me the gcode file and I will have a look (if possible the unprocessed gcode file so I can test on the exact file)

drsandytoes, 2020-07-23:
It's attached. Looking at some gcode exported from Canvas, I'm pretty sure the splice settings are getting generated incorrectly here as well.

[Raiders-Shield_0.2mm_PLA_MK3S_1h26m.gcode.zip](https://github.com/tomvandeneede/p2pp/files/4963929/Raiders-Shield_0.2mm_PLA_MK3S_1h26m.gcode.zip)

tomvandeneede, 2020-07-23:
Put a fix in the dev branch of GitHub.
can you please let me know if all ok now.   Checked with your file and seems the settings are taken from the table correctly now.

drsandytoes, 2020-07-23:
That fixes the parsing of the overrides. But it's generating this change table:

  Change instruction: D41 D0002 D0000 D0001
  Change instruction: D24 D0002 D0000 D0001
  Change instruction: D14 D0002 D0000 D0001

But D2 isn't even used; only D1 D3 and D4 are used in this file. I expected D34 and D43 should have the value 000 (HBPLA to PRPLA), and D13 and D31 should have the value 201.

tomvandeneede, 2020-07-23:
look at the O25 command

there the filament numbers are defined O25 D3FFFFFFWhitePRMTPLA D0 D2000000BlackHBPLA D4666666DimGrayPRPLA

in this case input 1= D3  2=unused   3=D2 and 4=D4

this comes from the parsing and due to the possibility to use more than 4 colors (best test) it is never renumbered to avoid issues
So impossible to tell from the extract below, but could be perfectly fine

drsandytoes, 2020-07-23:
Ah, thanks. I think this fixed the issue then.

closed 2020-07-23 by drsandytoes