tomvandeneede/p2pp #64: Z-move into print when using FULLPURGEREDUCTION

DwieDave, 2020-07-31:
Hello,
I recently discovered the FULLPURGEREDUCTION feature 
and im really happy with the purge tower result.
But when using the feature the print head moves in z-direction into the print 
(leaving an unpleasant hole) before performing the purge on a lower level.
Am I missing a setting or is that a bug or maybe is my print too near 
to the purge tower?

Attached my P2PP settings in the post processing gcode section:
```
;Palette 2 Configuration 
;P2PP PRINTERPROFILE={{real_printer_profile_id_from_canvas_here}}
;P2PP SPLICEOFFSET=40
;P2PP MINSTARTSPLICE=100
;P2PP MINSPLICE=70

; MATERIAL SPLICING INFORMATION
;P2PP MATERIAL_DEFAULT_0_0_0
;P2PP MATERIAL_PLA_PLA_0_0_0
;P2PP MATERIAL_PLA_PETG_0_0_0
;P2PP MATERIAL_PETG_PLA_0_0_0
;P2PP MATERIAL_PETG_PETG_0_0_0
;P2PP FULLPURGEREDUCTION
```

Best regards

tomvandeneede, 2020-07-31:
Can you please send me a copy of the unprocessed code so I can reproduce the processing

DwieDave, 2020-07-31:
Do you mean the GCode that will be handed to p2pp?

tomvandeneede, 2020-07-31:
yes please, gcode before it is processed by P2PP (just temporarily mess up the name of the script in the post processing setting, you will get an error but an output file will be generated

DwieDave, 2020-07-31:
[lcdknob.gcode.zip](https://github.com/tomvandeneede/p2pp/files/5006587/lcdknob.gcode.zip)
This is the file generated by PrusaSlicer 2.2.0 on macOS before post processing with P2PP.

tomvandeneede, 2020-07-31:
Could you do me a favour and print with variable layer height set off?

DwieDave, 2020-07-31:
Yes, of course I'm trying it now and report back; Thanks for your fast replies!

DwieDave, 2020-07-31:
That was apparently the problem:
Without variable layer height the z-movement did not happen again! :)

tomvandeneede, 2020-07-31:
I have meanwhile updated the code and added a warning for variable check compatibility

maybe in a later release I will try to fins a fix to allow both

DwieDave, 2020-07-31:
Sounds good; Thanks for your help and the great tool!

closed 2020-07-31 by DwieDave