tomvandeneede/p2pp #65: Total Filament Lengths

audiorazor, 2020-08-10:
Would it be possible to add something in the post processor that would add up total length (and maybe add an estimated weight) of each input of filament?   I know it breaks it down by input and length but it would be nice to see a total before starting a print so I know if I have enough.  

[image]

This is the info I am speaking of.

audiorazor, 2020-08-10:
sorry, just found that you already did this.... ill delete.

audiorazor, 2020-08-10:
sorry reopening, I see that the information isnt transferred to the g code file.     Ultimately what  I want to do is have a filament manager plugin be able to take that information so I can track spool usage.  
[image]

this information.

tomvandeneede, 2020-08-10:
Hi

not sure if this is one of the core things P2PP should do.
It technically could be done but it increase the dependency on parsing PrusaSlicer information… more dependency means more risk on Prusaslicer updates, without really adding any functional value.

I just checked the print info from quite a few users and the cost/density information is rarely filled.
it seems even prusaslicer does not release these numbers in their files (which would obviously not be correct after P2PP splicing)

I will see in the future if there is room for this but I cannot promise anything at this point..

audiorazor, 2020-08-10:
No problem.  I think we might have found a work around without having to
modify your p2pp output files.

closed 2020-09-19 by audiorazor