tomvandeneede/p2pp #67: Are material splice settings offsets to Palette or absolute?

pushc6, 2020-08-14:
I have "endurance" mode on on my Palette 2. Using splice tuning I found an offset of 1/0/0 made for the best pla->pla splices. If I set ;P2PP MATERIAL_PLA_PLA_1_0_0 will that _offset_ from endurance mode, or will it set the splice to 1/0/0? I just need to know whether I need to add the default "endurance" mode settings (2,-3,2) into my start g-code, so it'd be PLA_1_-3_2 instead of PLA_1_0_0. Thanks!

tomvandeneede, 2020-08-14:
Ryan,

good question… I have never used endurance mode…I will try to figure out

pushc6, 2020-08-14:
Thanks, I'm trying to tinker with it too, but thought I'd ask to see if you knew off the top of your head!

tomvandeneede, 2020-08-14:
just did a quick read, from what I see endurance mode is an offset of 2,-3,2 that is added to the supplied parameters… so if you start with 1_0_0 the actual value used will be 3,-3,2
the text around endurance is rather unclear on this subject, but I guess what they mean it you redo splice tuning in endurance mode and then see where this lands you and you use that new offset for your prints

closed 2020-09-04 by pushc6