tomvandeneede/p2pp #69: Feature Request: Splice testing mode

drsandytoes, 2020-08-26:
Feature request for a special splice testing (preflight) mode. In a special mode, it would be handy to read in the actual gcode file for a print, but instead of outputting the actual print file, output something that just generates a series of minimum-length splices that tests every unique transition in the actual print. Ideally, you'd never have to feed that length of filament to the print at all; it looks like if you remove the cover from the filament buffer on the palette, it never thinks it has filled the tube, and so keeps generating filament until you stop it. I've been doing something similar by manually hand editing the first few transitions in the O30 section to try out the different combinations, but I'm stuck with whatever splice lengths are already there, and I need to cancel the "print" manually when it reaches the end of my test pattern.

A similar tool could be useful for initial splice testing if it was possible to print a series of splices with varying splice parameters for each splice. For example, take two inputs, and create spliced output with varying degrees of heat, compression, and cooling at each minimum-length splice.

tomvandeneede, 2020-08-26:
will hev to think about this ons.  shouls be feasible bus cinfignwould bensort of cumbersome  will try to think of a way

tomvandeneede, 2020-12-20:
though potentially interesting,  it is not an easy thing to integrate into the tool at this point.   I will put this on the developmant list but will close it as an issue.   After the new version fully picks up I may integrate this

closed 2020-12-20 by tomvandeneede