tomvandeneede/p2pp #70: Absolute Extruder and Toolchange Parameters

pushc6, 2020-09-04:
So I have an mk3s with a palette 2. I was wondering if I should be using absolute extruder? The explanation kind of makes it sound like I should, but I'm still not abundantly clear it's impact or how it may affect pings on multiple small slices.

Second, in a p2pp user manual I found it has a section to set tool change parameters to all 0. Is this still necessary, or since it's not on the GitHub was it deemed not to be? Thanks!

tomvandeneede, 2020-09-04:
Impact should be minimal.  Relative extruder has the disadvantage that each extrusion length is rounded and rounding errors potentiallyy could build up
It all depends on how they are generated really

In absolute mode the counter is increased and the rounding is also sone but the overall affect is not cumulative so should give issues

I believe both of them work just fine…. I am quite sure you will maybe get q few mm difference qt the end but nobody will ever see this

pushc6, 2020-09-04:
What about the tool change parameters? We switched how our extruders work on our MK3s from the stepper directly extruding to a pulley box, where a smaller stepper rotates a larger pulley that drives extrusion. As part of that we set e-steps to E16 from E32 and number of steps to 490 from 280. We are seeing some bizarre results in our prints with color shifting layers and ping fluctuations. My prints aren't seeing any signs of extrusion issues, was just wondering if you had any ideas based on your knowledge of the palette\printer interfaces.

tomvandeneede, 2020-09-04:
so if i understand it correctly you went to 32 step microstepping and then reduced the number of steps?

I know that on some devices the micro stepping leads to less precise functioning.   I understood E16 was the sweet spot  and the chips would do 256microstepping in the background.
this however should result in local deviations sall uneven surfaces etc but not really on a global scale

Color shifting should not be an issue whenen the extruder remains properly calibrated.

What doe the pings look like?  do you see an increase/decrease (steady) of the pings that could indicate over or under extrusion against the values defined by the slicer

regards

tomvandeneede, 2020-12-20:
closing pending no feedback

closed 2020-12-20 by tomvandeneede