tomvandeneede/p2pp #72: Maybe an issue with newest p2pp?  Not sure

audiorazor, 2020-09-19:
I have been troubleshooting my printer and palette for day and days now.   I'm getting wild variations in my pings suddenly.   I have been using p2pp for over a year now with no issues and my start code hasnt changed.  I'm on a MK3S.  But lately my pings are all over the place and its putting colors where they dont belong.  Today I finally broke down and sliced the exact same file in Canvas and the pings are steady as a rock.   I attached the pings from the exact same print.   One sliced with p2pp and PS and the other with Canvas.  I'm going to attach the g code files also and hoping that you can either see that there is something wrong or point me in a direction because I am out of ideas.  

[3C_Lizard_1.txt](https://github.com/tomvandeneede/p2pp/files/5250003/3C_Lizard_1.txt)
[3C-Lizard.txt](https://github.com/tomvandeneede/p2pp/files/5250005/3C-Lizard.txt)
[3C_Lizard_1 P2PP.mcf.txt](https://github.com/tomvandeneede/p2pp/files/5250010/3C_Lizard_1.P2PP.mcf.txt)
[3C-Lizard CANVAS.mcf.txt](https://github.com/tomvandeneede/p2pp/files/5250011/3C-Lizard.CANVAS.mcf.txt)

tomvandeneede, 2020-09-19:
Hi 

Did you do any update on P2PP or PrusaSlicer.   Will need some more material to work with to understand where the problem comae from

audiorazor, 2020-09-19:
The only thing I can think that I added was the few lines of code that you
added.  The material temp change one and the version check.   I always keep
my dev version up to date and was wondering if it was something that got
accidentally changed or if it was me.

I'm about to roll back two versions of p2pp to see if the issue persists or
goes away.    What other info do you need from me?

audiorazor, 2020-09-19:
Rolling back 2 versions of p2pp on my mac did not resolve the issue.  I'm honestly at a loss as to what is going on.  Been troubleshooting this for days and days now.

audiorazor, 2020-09-20:
Turns out I am in idiot.   Thanks for playing along!

closed 2020-09-20 by audiorazor

tomvandeneede, 2020-09-20:
Hi,

don’t worry, I get many issues that are hard to find hence trying to get more info (plus it was a beautiful day yesterday so was not in most of the time hence my lack of response)
I try to test changes very extensively and feedback like yours is really helpful to keep P2PP as stable as possible

I hope all is printing fine now… if not let me know happy to help