tomvandeneede/p2pp #73: BigBrain3D RPM - "Reset correct Z height" does not work correctly

deepthought-64, 2020-09-24:
I am using the retractable purge mechanism from BigBrain3D (i know support is stil alpha).

During my first print I noticed, that the restoration of the Z-Height after making the blobs does not work correctly. 
In my example I had a tool change in the first layer (z=0.2mm) and for making the blobs the printer rises to z=20 but after making the blobs, the printer returns not to the correct height but to z=0.8mm.

It looks like the parser does not recognize what the correct "old z" is.

Here an excerpt from the G-Code file just before the first tool change:

```
...
G1 X177.570 Y100.996 E0.02940
G1 X177.437 Y100.678 F10800.0
G1 Z0.800
G1 F8640.0
G1 X177.350 Y100.690 E-0.08705
G1 X177.391 Y100.039 E-0.15053
G1 X177.456 Y99.443 E-0.13864
G1 X177.548 Y98.849 E-0.13864
G1 X177.665 Y98.260 E-0.13864
G1 X177.775 Y97.812 E-0.10652
G1 E-0.04000 F2100.0
;-- P2PP -- removed [tool unload] - G1 X143.609 Y252.102 F10800.0
;-- P2PP -- removed [tool unload] - G1 Z0.200
;-- P2PP -- removed [tool unload] - G1 E0.80000 F2100.0
;--------------------
; CP TOOLCHANGE START
; toolchange #1
; material : PLA -> PLA
;--------------------
...
```

For me it looks like the P2PP uncomments the line with the correct z-height `;-- P2PP -- removed [tool unload] - G1 Z0.200` and uses the previous `G1 Z0.800` as the z-height it should return to after making the blobs.

tomvandeneede, 2020-09-24:
Can you please send me the unprocessed and processed gcode for your project so I can have  a look.

thanks!

deepthought-64, 2020-09-24:
Hi
Sorry, of course I forgot.
Unfortunately this is a proprietary project - I hope It helps if I attach the first 30k lines of the input and output files. 
There are 3 tool changes in there. I hope that is sufficient. 

I tried to reproduce the problem with just a simple project made of 2 boxes, but the bug does not appear then. 
If I can provide any more info, just let me know.

Thanks for your support!

tomvandeneede, 2020-09-24:
well

sort of does not help as I need to be able to test
aby chance you can reproduce the problem on a non-prop model and send me those files?
browsing through 30K lines without the help of a debugger to stop code generation at the right point is kind of cumbersome prospect

regards

deepthought-64, 2020-09-24:
Yeah, I understand. Let me see if i can reproduce that on another model. Will get back to you as soon as i have something.

tomvandeneede, 2020-12-20:
closed since no further feedback on issue

closed 2020-12-20 by tomvandeneede