tomvandeneede/p2pp #76: First Layer issues - Wrong Z height in the conversion?

jmogl, 2020-11-09:
I have not been successful printing multi-extruder gear bearing from Thingiverse. Portions of the outer ring gear will not adhere to the plate since the Z-height appears to be off. I can get it to print on Mosaic Canvas, so it does not appear to be a physical issue with the printer.

Link to the model:
https://www.thingiverse.com/thing:4577195

I would appreciate if you can track down what is not converting correctly. The outer section of the purge tower is created along with the skirt. The portion that fails is the initial part of the teeth for the main outer circular gear, which are not contiguous. After it goes past these, the remaining part of the gear starts to print normally.

Gcode attached.
[Herringbone_Planetary_Gear_0.2mm_PLA_MK3S_5h39m.mcf.zip](https://github.com/tomvandeneede/p2pp/files/5508299/Herringbone_Planetary_Gear_0.2mm_PLA_MK3S_5h39m.mcf.zip)

jmogl, 2020-11-09:
In the picture below, the failed portion was the first part of the gear. The circle and teeth that are shown came after and appear to go down normally.

[image]

tomvandeneede, 2020-11-09:
Hi Jeff,

can you be more specific,

what version of P2PP are you using
please PM me in Facebook messenger the GCODE files for both the processed and the unprocessed project

I’m sure we can get this fixed

jmogl, 2020-11-09:
Tom,

I appreciate you looking into this!

P2PP version 4.16.000. The original file was downloaded from Thingiverse with the link above along with the print gcode. I uploaded the project file in this message.

[Herringbone_Planetary_Gear.zip](https://github.com/tomvandeneede/p2pp/files/5508430/Herringbone_Planetary_Gear.zip)

I no longer use Facebook, so let me know if you need me to upload any other files here.

Thanks!

tomvandeneede, 2020-11-09:
Hi Jeff,

I really need the gcode files you produced (both before and after running P2PP)
You can send them through Dropbox/Wetransfer/Onedrive as well.
Also which version of PrusaSlicer are you using?

Regards

tomvandeneede, 2020-11-09:
Sorry,

get the file now… for some reason it was blocked by the OS, I will have a look

jmogl, 2020-11-09:
PrusaSlicer-2.2.0. Let me know if you need anything else. BTW, I tried adjusting the Z-height offset on the MK3S and it was about 0.1mm to get the first layer to adhere fully, but then there were other artifacts since the print head was too close to the plate.

tomvandeneede, 2020-11-09:
Jeff,

from initial investigation all looks ok:

you see the printer increase Z to 0.6 to move o the brim location and then it returns to 0.2 to print the brim.
After that if does a similar up and down positioning move  and starts printing the contour of the gear

retractions are also ok so no missing filament

if the print is coming off the bed it might be a z-height issue or an adhesion issue.  The great structure has a lot of turns in it ans once it does nog stick it tends to continue not doing that

From a gcode perspective all looks fine though, all is printing at the proper height

G1 E-0.30000 F2100.0
G1 Z0.600 F10800.0
G1 Z0.200
;AFTER_LAYER_CHANGE
;0.2
;LAYER 0
;LAYERHEIGHT 0.2
;-- P2PP -- removed [Double Retract] - G1 E-3.70000 F2100.0
G1 Z0.600 F10800.0
G1 X234.216 Y77.088
G1 Z0.200
G1 E0.30000 F2100.0
;-------------------------------------
; CP WIPE TOWER FIRST LAYER BRIM START
G1 X234.216 Y138.788 E3.51750 F2400
G1 X231.673 Y139.245 F7000
…..
G1 X233.044 E2.66130
G1 X231.216 Y138.788 F7000
G1 X171.216
G1 X231.216
; CP WIPE TOWER FIRST LAYER BRIM END
;-----------------------------------

G1 E-0.30000 F2100.0
G1 Z0.600 F10800.0
G1 X234.404 Y144.166
G1 Z0.200
G1 E0.30000 F2100.0
M204 S1000
G1 F1200.0
G1 X233.282 Y144.353 E0.03568
G1 X168.873 Y144.342 E2.01950
G1 X126.263 Y135.231 E1.36620
G1 X124.554 Y134.847 E0.05491
G1 X122.779 Y134.331 E0.05798
G1 X121.044 Y133.706 E0.05783
G1 X119.261 Y132.928 E0.06097
G1 X117.716 Y132.139 E0.05440
G1 X116.224 Y131.258 E0.05434
G1 X114.694 Y130.227 E0.05783
G1 X113.157 Y129.042 E0.06087

jmogl, 2020-11-09:
Not sure what the issue is then. 

After it prints the outline of the purge tower and the skirt, it starts to print small parts of the gear that do not adhere at all. Plastic starts to collect by the nozzle and just gets pushed around. Then it starts the zig zag of the gear teeth which does adhere. I wouldn't expect it to suddenly start adhering in the same general space if it was a physical issue. I used the nylock mod to level the plate. Running the same model right now in Canvas and it is 86% complete so far without glue which also indicates it is not hardware.

PrusaSlicer and P2PP fails exactly at the same point each time I tried to run it unless I force the Z-offset down with or without glue. But as mentioned above, moving the z-offset doesn't work for the rest of the print. Any idea what I should try next if the g-code looks correct?

tomvandeneede, 2020-11-09:
Can you try to slice the model in a single color in PrusaSlicer an see what it does then?

I have to go by the GCODE file… it does not show anything special.  All  nicely printed at 0.2mm height…. so if there is an issue it is not the p2pp processing as far as I can see, potentially still something in PS, though I would not know what it is….

if canvas prints ok then your z-offset should be fine

P2PP itself does little processing to this file.   It just removes MMU specific code and inserts the required ping codes into your file.

hope you can isolate the issue...

jmogl, 2020-11-09:
Prints perfectly with single color from Prusa Slicer. I stopped it after a couple of layers since I was testing prior to Canvas print. So the issue is somewhere with the palette. Just not sure why it is working with Canvas and not PrusaSlicer/P2PP.

tomvandeneede, 2020-11-09:
can you send me the gcode in prusa with single color that prints ok?

jmogl, 2020-11-09:
G-code attached. Note that this is the single STL version where the multi-filament print had the individual pieces broken out to make the color changes. 

[G-code_GearBearingTest.zip](https://github.com/tomvandeneede/p2pp/files/5508650/G-code_GearBearingTest.zip)

tomvandeneede, 2020-11-09:
only difference I see is in the retract lengths and in the Filament start gcode (M900 command)

rest is very much comparable, same heights etc

left is single color

could be any of these but not sure really why except maybe for the last one this would make much of a diff

jmogl, 2020-11-09:
I don't think the model has a corruption issue since this the exact file just printed successfully in Canvas. It didn't print as tight of a tolerance as the single color default Prusa gcode version that I had done earlier, But it worked:
[image]

tomvandeneede, 2020-11-09:
Hey Jeff

spent some time this evening looking into your file and there is really nothing that I can come up with that would explain

I notices some differences in the setting, like retractions, wiping etc but normally that shouldn’t be that much of an issue
all is printed at pretty much the same speed same extrusions, ….

No doubt the model is ok, and even if it were, it should not be the cause of not adhering.

What I also checked is the involvement of P2PP in the first part of the code and there is hardly any..  The header is inserted at the beginning but that is inert gcode, the printer will  not process it.
There are some retractions that are removed but they should not matter that much as the retractions are offset by the un-retracts in the later part of the code.  
you can just increase the amount of retractions to 0.4 instead of 0.3 to match the setting in your single color print.

If you can provide me with an unprocessed file and the a processed file from p2pp I can take a deeper look at the original file… but I expect no to find much.

Concerning the comparison to Canvas is difficult as this is an entirely different slicer.   

So in summary:  P2PP is doing what it is expected to do from single gcode file investigation perspective.  I don’t see obvious mistakes in either the speed of extraction nor the material volumes…

It is likely not the model, probably also not P2PP so remains the Z-heigh of your printer (did you calibrate it recently?), some minor setting differences causing this behaviour.

This is as much as I can see with the files at hand.

Regards
Tom

jmogl, 2020-11-09:
Tom,

I had forced the Z height calibration down to get the gear bearing to adhere when using prusa slicer and P2PP, but it failed. The head was too close causing other issues. I reran the  height calibration to get it correct and Prusa Slicer with P2PP fails, but works fine with single Prusa slicer direct and Canvas. I am wondering if I messed up a setting somewhere. I won’t be able to do any more troubleshooting until next weekend. 

Thanks again for taking the time to look into this!

Jeff

skellatore, 2020-11-10:
I've had a look at the file too @tomvandeneede .. All looks fine. Based on the photo, I can see lint and/or dust. I suspect a dirty bed is the culprit. The purple looks like it's got some moisture problems or it's being printed too hot based on the photo - if you have a look at the bottom RHS of the purge tower, it's more obvious there.

Likely to be one or two things at play;
- Either the bed is dirty. Clean with soap & water, wipe with lint-free cloth with Isopropyl/denatured alcohol.
- Filament too hot and popping (steam). Lower the temperature by 5c or so.
- Bed isn't level well enough for ABL to compensate. A Nyloc nut mod (outside of the scope of anything p2pp) may help.

It's definitely _not_ P2PP related and looks to be a coincidence.

jmogl, 2020-11-10:
Tim,

Based on your review, please close the ticket. I will need to trouble shoot more at my end. I know it is not a level issue since I did the nylock mod and have the plate close to 0.05 mm. Which I think is the tolerance of the Pinda. I have no issues with adhesion when printing it with Canvas or single color on Prusa Slicer. It is really strange that the first part of the gear does not stick and then the rest does unless I force the level height down. I have Z height properly calibrated so I will need to look at the differences in Gcode with the Canvas version. 

I initially thought this was a conversion issue. Thank you very much for taking the time to look at this!

Jeff

jmogl, 2020-11-22:
Just a quick update. I have been traveling and didn't get a chance to run more tests until today. I ran a layer test print with and without P2PP and it worked perfectly. I had reset Prusa Slicer settings. The gear is printing now and the first layer printed fine. Not sure if it was a setting issue or just needed a deep clean. Confusing since it worked with Canvas. I should have done more testing at my end before opening a ticket. Thanks again for your help and developing P2PP!

closed 2020-11-22 by jmogl