tomvandeneede/p2pp #78: Missing layers in purge block

audiorazor, 2020-11-28:
Not sure if this is related to the newest p2pp or the 2.3 beta but I didnt have this issue until today.   The purge block is skipping a layer.   I will attach the g code and a picture for you to reference.   I'm going to see if I can figure out how to roll back my p2pp version to see if the issue goes away in the meantime.  
[Miles Morales.mcf.gcode.zip](https://github.com/tomvandeneede/p2pp/files/5610774/Miles.Morales.mcf.gcode.zip)
Its right when the first color change happens so only about 10 layers into the print.

tomvandeneede, 2020-11-28:
could you please send me the unprocessed file.
will need that to see where the problem is.

thanks

audiorazor, 2020-11-28:
Like the file without going through p2pp?

tomvandeneede, 2020-11-28:
yes please

audiorazor, 2020-11-28:
[same file no p2pp.gcode.zip](https://github.com/tomvandeneede/p2pp/files/5610789/same.file.no.p2pp.gcode.zip)

audiorazor, 2020-11-28:
Also it looks like the mac's version number was never updated.  still says 5.13

audiorazor, 2020-11-28:
I see you updated.  Do you think you figured out what it was?

tomvandeneede, 2020-11-28:
I hope so,  ran it again and the block was solid

audiorazor, 2020-11-28:
Ok.  I’ll give it another try.

audiorazor, 2020-11-28:
[image]
The purge block is solid again but now I am getting some weirdness on the edges.   Attaching photos to look at.   
[image]

tomvandeneede, 2020-11-28:
Photo’s did not come through

 

 <https://user-images.githubusercontent.com/14953137/100525605-1d350e00-3190-11eb-8b3e-b3aedeb4887c.jpg> 

 <https://user-images.githubusercontent.com/14953137/100525615-2f16b100-3190-11eb-996e-2b7fc4a905a8.jpg>

audiorazor, 2020-11-28:
[image]
[image]

audiorazor, 2020-11-28:
how about now?

tomvandeneede, 2020-11-28:
Still blank mail

audiorazor, 2020-11-28:
sent the photos directly to the email you have on the CC.   Hopefully that works.

tomvandeneede, 2020-11-28:
Ok got them now

Will look into this first thing in the morning

audiorazor, 2020-11-28:
Thank you.   Do you need any g code from me?

tomvandeneede, 2020-11-28:
Shoul be fine

audiorazor, 2020-11-29:
I see you updated again.  Am I good to try it again?

audiorazor, 2020-11-29:
Thank you.  All fixed.   I just tested on a new print.    Thanks for all the work you do.

closed 2020-11-29 by audiorazor

tomvandeneede, 2020-11-29:
Thanks for reporting back 

 

Happy printing.