tomvandeneede/p2pp #79: Soluble Supports Warning Message

kats-greg, 2020-12-11:
I am trying to set up a print using soluble support material, and when I send the gcode to the printer I get this error message from P2PP 

WARNING: No Algorithm defined for transitioning PLA to PVA. Using Default

Is there a setting/configuration I need to add?

Thanks

tomvandeneede, 2020-12-11:
Hi 

Would you mind sending me the first 200 lines of your gcode (including the P2PP lines)

thanks!

kats-greg, 2020-12-11:
Here you go.

[200Lines..mcf.gcode.txt](https://github.com/tomvandeneede/p2pp/files/5678157/200Lines.mcf.gcode.txt)

Thanks

tomvandeneede, 2020-12-11:
at first look you are missing the 

;P2PP MATERIAL_PLA_PVA_?_?_?

you need both PVA_PLA and PLA_PVA lines

line in your startup gcode

tomvandeneede, 2020-12-11:
add a line

;P2PP MATERIAL_PLA_PVA_0_0_0
and you should be fine, though the default is 0_0_0 so the current file would have printed just fine

kats-greg, 2020-12-11:
That did it.

Thanks

On 12/11/2020 3:03 AM, Tom Van den Eede wrote:
> ;P2PP MATERIAL_PLA_PVA_0_0_0

closed 2020-12-13 by tomvandeneede