tomvandeneede/p2pp #80: just upgraded to latest big sur macOS 11.1 (20C69)

wnc2, 2020-12-15:
[image]
[image]

tomvandeneede, 2020-12-15:
Hi

this error has been reported with the beta version of Big Sur before and now seems to resurface with the new upgrade.
I am running 11.0 and do not have any issues.   I will upgrade to 11.1 overnight and will see if the issues emerge.

I will try to figure out what you can do to make this go away as this is very annoying

regards

wnc2, 2020-12-15:
never had issues me too, only just after upgrade to 11.1

tomvandeneede, 2020-12-15:
sorry to hear that…. it is very new update so I expect a solution to be available pretty soon
specially since you are using the standard python delivered with the system

wnc2, 2020-12-15:
with very big effort, i (think i) have installed python 3.9.1, but i read 3.7.5 
[image]
but now i dont know what to do

tomvandeneede, 2020-12-15:
hmm it all has to do with the virtual environments I believe.
In the crash report you sent I saz 2.7.x which I think is the standard supplied

wnc2, 2020-12-15:
why i have installed python 3.9.1, i read instead 3.7.5 and i RUN 2.7?
[image]
[image]

audiorazor, 2020-12-16:
I'm having the same issue

tomvandeneede, 2020-12-16:
Did you also install 11.1?

audiorazor, 2020-12-16:
I did.

tomvandeneede, 2020-12-16:
same here since this morning.

just install python 3 and change the p2pp.command script as follows

#!/bin/sh
DIRECTORY=`dirname $0`

if [ $# -eq -0 ]
then
     python3 $DIRECTORY/P2PP.py
else
     python3 $DIRECTORY/P2PP.py -i "$1"
fi

this should do the trick 9at least worked on my machine on which I have python 3.7.0

audiorazor, 2020-12-16:
I tried that earlier but got a different crazy error.  I’ll post a link
when I wake up tomorrow.

wnc2, 2020-12-16:
this is my result when i run the new p2pp.command
[image]
[image]

audiorazor, 2020-12-16:
This is the same error I am getting now.

audiorazor, 2020-12-16:
[image]

tomvandeneede, 2020-12-16:
looks indeed like your python env is not setup for tk
I will need to do some further investigation to this

Very nice of apple to release stuff like this but guess they can do what they want… though I expect them to patch it some day

audiorazor, 2020-12-16:
yea, ive been googling for solutions all day.  

I found this but I dont know how to implement any of these things as I'm not a coder.  
https://stackoverflow.com/questions/5459444/tkinter-python-may-not-be-configured-for-tk

tomvandeneede, 2020-12-16:
do you have xcode installed?

seems python 3 is part of the xcode distribution

audiorazor, 2020-12-16:
I hadnt downloaded it because it was 11GB of data I didnt think I needed on my hard drive.   Guess ill download it now.

tomvandeneede, 2020-12-16:
fully agree this should not be necessary
peen or virtualenv could also bring a solution

tomvandeneede, 2020-12-16:
this might be useful but still have to try

https://opensource.com/article/19/5/python-3-default-mac

wnc2, 2020-12-16:
[image]

from: https://scriptingosx.com/2020/06/macos-11/ 

[image]

tomvandeneede, 2020-12-16:
well,

there is not much to do about that… but I am sure we can get this to a working solution

wnc2, 2020-12-16:
i found this:
1) there is an idle.app (something related to python 3.9.1) i launched it
2) i opened p2pp.py through idle.app
3) the initial P2PP setup works well, but then i have the same error running from prusaslicer

hope this helps
[image]

tomvandeneede, 2020-12-16:
Well when I run the app from within PyCharm I also don’t have the problem
I installed Python 3.9.1 and still I get this error when running from the command line.

I maybe should consider using Qt instead of TkInter for my screen
Should be a minor change, but will require installation of libraries by the users

wnc2, 2020-12-16:
tell me step by step what to do and i try.

BTW although i surely have python 3.9.1 installed, when python crashes during script inside prusaslicer i still read python 2.7

tomvandeneede, 2020-12-16:
I have no step by step thing for this… this is outside of my comfort zone at the moment
I only know Tkinter is the problem so I will see if there is an easy replacement for TKInter that will make this go away 

This is not really a p2pp problem but a lot bigger so I hope/assume others will be looking into this as well with a better python Tkinter background

wnc2, 2020-12-16:
after xcode install i have a brand new crash report......
[image]

tomvandeneede, 2020-12-16:
it stays the same
 
I installed 3.9.1 and ran the 2 command files and I now get 3.9.1 but still the same error
trying to get tkinter updated now

tomvandeneede, 2020-12-16:
The python 3.9.1 still uses Tcl/Tk 8.5 apparently.   trying to fins a way to link the correct framework… looks like non-trivial exercise

tomvandeneede, 2020-12-16:
some more info:

https://www.python.org/download/mac/tcltk/#built-in-8-6-8

audiorazor, 2020-12-16:
Wish I never updated my OS now.   What a pain in the ass.

tomvandeneede, 2020-12-17:
don’t panic yet…we’ll find a solution

benbusser, 2020-12-17:
Tom that would be great. Just bought the Palette 2s and everyone recommends to use PrusaSlicer and now I bumped into same issue as wnc2. It was already a steep way to get p2pp running.....
Thanks for the great work, hope to be able to use it soon

tomvandeneede, 2020-12-17:
can you please try the following

1/ uninstall any python 3 you have installed 
2/ download and install python3 from python.prg (I took version 3.9.1)
3/ use the following for the p2pp.sommand file

#!/bin/sh
DIRECTORY=`dirname $0`

if [ $# -eq -0 ]
then
     python3 $DIRECTORY/P2PP.py
else
     python3 $DIRECTORY/P2PP.py -i "$1"
fi

audiorazor, 2020-12-17:
I still get the same error.  

Post-processing script /Users/nicholassnyder/Documents/p2pp_dist/p2pp.command on file /Users/nicholassnyder/Desktop/Black Parts Sippy Box Cat.mcf.gcode failed.
Error code: 1
Output:
Traceback (most recent call last):
File "/Users/nicholassnyder/Documents/p2pp_dist/p2pp/gui.py", line 12, in <module>
import Tkinter as tkinter
ModuleNotFoundError: No module named 'Tkinter'
During handling of the above exception, another exception occurred:
Traceback (most recent call last):
File "/Users/nicholassnyder/Documents/p2pp_dist/P2PP.py", line 18, in <module>
import p2pp.gui as gui
File "/Users/nicholassnyder/Documents/p2pp_dist/p2pp/gui.py", line 17, in <module>
import tkinter
File "/Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.7/lib/python3.7/tkinter/__init__.py", line 36, in <module>
import _tkinter # If this fails your Python may not be configured for Tk
ModuleNotFoundError: No module named '_tkinter'

tomvandeneede, 2020-12-17:
which python did you setup?

audiorazor, 2020-12-17:
3.9.1 via python.org.

I made it this far yesterday

tomvandeneede, 2020-12-17:
which version did you install?
can you run the following command

python -m test.pythoninfo

potentially like python3 -m test.pythoninfo (with the version number)

the first will return normally python 2.7.16 info from the python setup with your Mac
the second should get you python3 info, there should be Tcl/TK framework versions in tha list
if not TCL/YK is not installer
8.5 is wrong version
8.6 seems fine

you can also install 8.6 from ActiveTCL (I did that but not sure if it was needed)

audiorazor, 2020-12-17:
It says 8.6 is installed.

This is working for you on python3?

benbusser, 2020-12-17:
Tom,

I did this already
Checked the python version as well

python -m test.pythoninfo
Python debug information
========================

CC.version: Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Py_DEBUG: No (sys.gettotalrefcount() missing)
_decimal.__libmpdec_version__: 2.5.0
builtins.float.double_format: IEEE, little-endian
builtins.float.float_format: IEEE, little-endian
config[_config_init]: 2
config[_init_main]: 1
config[_install_importlib]: 1
config[_isolated_interpreter]: 0
config[_orig_argv]: ['/Library/Frameworks/Python.framework/Versions/3.9/Resources/Python.app/Contents/MacOS/Python', '-m', 'test.pythoninfo']
config[_use_peg_parser]: 1
config[argv]: ['-m']
config[base_exec_prefix]: '/Library/Frameworks/Python.framework/Versions/3.9'
config[base_executable]: '/usr/local/bin/python'
config[base_prefix]: '/Library/Frameworks/Python.framework/Versions/3.9'
config[buffered_stdio]: 1
config[bytes_warning]: 0
config[check_hash_pycs_mode]: 'default'
config[configure_c_stdio]: 1
config[dev_mode]: 0
config[dump_refs]: 0
config[exec_prefix]: '/Library/Frameworks/Python.framework/Versions/3.9'
config[executable]: '/usr/local/bin/python'
config[faulthandler]: 0
config[filesystem_encoding]: 'utf-8'
config[filesystem_errors]: 'surrogateescape'
config[hash_seed]: 0
config[home]: None
config[import_time]: 0
config[inspect]: 0
config[install_signal_handlers]: 1
config[interactive]: 0
config[isolated]: 0
config[malloc_stats]: 0
config[module_search_paths]: ['/Library/Frameworks/Python.framework/Versions/3.9/lib/python39.zip', '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9', '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/lib-dynload']
config[optimization_level]: 0
config[parse_argv]: 1
config[parser_debug]: 0
config[pathconfig_warnings]: 1
config[platlibdir]: 'lib'
config[prefix]: '/Library/Frameworks/Python.framework/Versions/3.9'
config[program_name]: '/usr/local/bin/python'
config[pycache_prefix]: None
config[pythonpath_env]: None
config[quiet]: 0
config[run_command]: None
config[run_filename]: None
config[run_module]: 'test.pythoninfo'
config[show_ref_count]: 0
config[site_import]: 1
config[skip_source_first_line]: 0
config[stdio_encoding]: 'utf-8'
config[stdio_errors]: 'strict'
config[tracemalloc]: 0
config[use_environment]: 1
config[use_hash_seed]: 0
config[user_site_directory]: 1
config[verbose]: 0
config[warnoptions]: []
config[write_bytecode]: 1
config[xoptions]: []
[... 144 lines cut ...]
test_socket.HAVE_SOCKET_BLUETOOTH: False
test_socket.HAVE_SOCKET_CAN: False
test_socket.HAVE_SOCKET_CAN_ISOTP: False
test_socket.HAVE_SOCKET_CAN_J1939: False
test_socket.HAVE_SOCKET_QIPCRTR: False
test_socket.HAVE_SOCKET_RDS: False
test_socket.HAVE_SOCKET_UDPLITE: False
test_socket.HAVE_SOCKET_VSOCK: False
test_support._is_gui_available: True
test_support.python_is_optimized: True
time.altzone: -7200
time.daylight: 1
time.get_clock_info(monotonic): namespace(implementation='mach_absolute_time()', monotonic=True, adjustable=False, resolution=1e-09)
time.get_clock_info(perf_counter): namespace(implementation='mach_absolute_time()', monotonic=True, adjustable=False, resolution=1e-09)
time.get_clock_info(process_time): namespace(implementation='clock_gettime(CLOCK_PROCESS_CPUTIME_ID)', monotonic=True, adjustable=False, resolution=1.0000000000000002e-06)
time.get_clock_info(thread_time): namespace(implementation='clock_gettime(CLOCK_THREAD_CPUTIME_ID)', monotonic=True, adjustable=False, resolution=1e-09)
time.get_clock_info(time): namespace(implementation='clock_gettime(CLOCK_REALTIME)', monotonic=False, adjustable=True, resolution=1.0000000000000002e-06)
time.time: 1608221787.112897
time.timezone: -3600
time.tzname: ('CET', 'CEST')
tkinter.TCL_VERSION: 8.6
tkinter.TK_VERSION: 8.6
tkinter.info_patchlevel: 8.6.10
zlib.ZLIB_RUNTIME_VERSION: 1.2.11
zlib.ZLIB_VERSION: 1.2.11

tomvandeneede, 2020-12-17:
yes it does

tomvandeneede, 2020-12-17:
this should work

benbusser, 2020-12-17:
well it does not
When sending to octoprint I get this:

Post-processing script /Applications/p2pp_dist/p2pp.command on file /var/folders/jj/q6qhrtsj16752stcjkv20gd00000gn/T/.PrusaSlicer.upload.4783-fef5-f9aa-4d61 failed.
Error code: 134
Output:
macOS 11 or later required !
/Applications/p2pp_dist/p2pp.command: line 9: 21586 Abort trap: 6 python3 $DIRECTORY/P2PP.py -i "$1"

and Python crashes:
Process:               Python [21627]
Path:                  /Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/Resources/Python.app/Contents/MacOS/Python
Identifier:            com.apple.python3
Version:               3.8.2 (3.8.2)
Build Info:            python3-73040006000000~117
Code Type:             X86-64 (Native)
Parent Process:        ??? [21625]
Responsible:           PrusaSlicer [21576]
User ID:               501

PlugIn Path:             /Library/Developer/CommandLineTools/Library/Frameworks/Python3.framework/Versions/3.8/lib/python3.8/lib-dynload/_tkinter.cpython-38-darwin.so
PlugIn Identifier:       _tkinter.cpython-38-darwin.so
PlugIn Version:          ??? (73.40.6)

Date/Time:             2020-12-17 17:32:41.588 +0100
OS Version:            macOS 11.1 (20C69)
Report Version:        12
Anonymous UUID:        241D3085-495E-DD84-EF60-DBAA610894B3

Sleep/Wake UUID:       8BED0CAF-0D2D-4DA1-989B-FA855122B379

Time Awake Since Boot: 53000 seconds
Time Since Wake:       2500 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

tomvandeneede, 2020-12-17:
which version did you install from python.org?

benbusser, 2020-12-17:
Good question, how do I find out?

tomvandeneede, 2020-12-17:
what I see is that the version used when running command file is not the same as the command line version

can you create a test.command like this

#!/bin/sh
python3 -version
python3 -m test.pythoninfo

then do the same in a terminal.

python 2.7.16 (the default python that comes with the mac) does not work so you will need to default to python3 which is mainstream since quite some time now

tomvandeneede, 2020-12-17:
I had to remove the home-brew installations I did to make things work

all brew bottles seem to contain references to the 8.5 framework… the python.org has more up to date 8.6.9

benbusser, 2020-12-17:
This what I see when I run test.command
benbusser@iMac-van-Ben ~ % /Applications/p2pp_dist/test.command ; exit;
Unknown option: -e
usage: /Library/Frameworks/Python.framework/Versions/3.9/bin/python3 [option] ... [-c cmd | -m mod | file | -] [arg] ...
Try `python -h' for more information.
Python debug information
========================

CC.version: Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/c++/4.2.1
Py_DEBUG: No (sys.gettotalrefcount() missing)
_decimal.__libmpdec_version__: 2.5.0
builtins.float.double_format: IEEE, little-endian
builtins.float.float_format: IEEE, little-endian
config[_config_init]: 2
config[_init_main]: 1
config[_install_importlib]: 1
config[_isolated_interpreter]: 0
config[_orig_argv]: ['/Library/Frameworks/Python.framework/Versions/3.9/Resources/Python.app/Contents/MacOS/Python', '-m', 'test.pythoninfo']
config[_use_peg_parser]: 1
config[argv]: ['-m']
config[base_exec_prefix]: '/Library/Frameworks/Python.framework/Versions/3.9'
config[base_executable]: '/Library/Frameworks/Python.framework/Versions/3.9/bin/python3'
config[base_prefix]: '/Library/Frameworks/Python.framework/Versions/3.9'
config[buffered_stdio]: 1
config[bytes_warning]: 0
config[check_hash_pycs_mode]: 'default'
config[configure_c_stdio]: 1
config[dev_mode]: 0
config[dump_refs]: 0
config[exec_prefix]: '/Library/Frameworks/Python.framework/Versions/3.9'
config[executable]: '/Library/Frameworks/Python.framework/Versions/3.9/bin/python3'
config[faulthandler]: 0
config[filesystem_encoding]: 'utf-8'
config[filesystem_errors]: 'surrogateescape'
config[hash_seed]: 0
config[home]: None
config[import_time]: 0
config[inspect]: 0
config[install_signal_handlers]: 1
config[interactive]: 0
config[isolated]: 0
config[malloc_stats]: 0
config[module_search_paths]: ['/Library/Frameworks/Python.framework/Versions/3.9/lib/python39.zip', '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9', '/Library/Frameworks/Python.framework/Versions/3.9/lib/python3.9/lib-dynload']
config[optimization_level]: 0
config[parse_argv]: 1
config[parser_debug]: 0
config[pathconfig_warnings]: 1
config[platlibdir]: 'lib'
config[prefix]: '/Library/Frameworks/Python.framework/Versions/3.9'
config[program_name]: '/Library/Frameworks/Python.framework/Versions/3.9/bin/python3'
config[pycache_prefix]: None
config[pythonpath_env]: None
config[quiet]: 0
config[run_command]: None
config[run_filename]: None
config[run_module]: 'test.pythoninfo'
config[show_ref_count]: 0
config[site_import]: 1
config[skip_source_first_line]: 0
config[stdio_encoding]: 'utf-8'
config[stdio_errors]: 'strict'
config[tracemalloc]: 0
config[use_environment]: 1
config[use_hash_seed]: 0
config[user_site_directory]: 1
config[verbose]: 0
config[warnoptions]: []
config[write_bytecode]: 1
config[xoptions]: []
datetime.datetime.now: 2020-12-17 17:45:36.569485
[... 151 lines cut ...]
test_support._is_gui_available: True
test_support.python_is_optimized: True
time.altzone: -7200
time.daylight: 1
time.get_clock_info(monotonic): namespace(implementation='mach_absolute_time()', monotonic=True, adjustable=False, resolution=1e-09)
time.get_clock_info(perf_counter): namespace(implementation='mach_absolute_time()', monotonic=True, adjustable=False, resolution=1e-09)
time.get_clock_info(process_time): namespace(implementation='clock_gettime(CLOCK_PROCESS_CPUTIME_ID)', monotonic=True, adjustable=False, resolution=1.0000000000000002e-06)
time.get_clock_info(thread_time): namespace(implementation='clock_gettime(CLOCK_THREAD_CPUTIME_ID)', monotonic=True, adjustable=False, resolution=1e-09)
time.get_clock_info(time): namespace(implementation='clock_gettime(CLOCK_REALTIME)', monotonic=False, adjustable=True, resolution=1.0000000000000002e-06)
time.time: 1608223536.6304388
time.timezone: -3600
time.tzname: ('CET', 'CEST')
tkinter.TCL_VERSION: 8.6
tkinter.TK_VERSION: 8.6
tkinter.info_patchlevel: 8.6.10
zlib.ZLIB_RUNTIME_VERSION: 1.2.11
zlib.ZLIB_VERSION: 1.2.11

Saving session...
...copying shared history...
...saving history...truncating history files...
...completed.
Deleting expired sessions...none found.

[Proces voltooid]

tomvandeneede, 2020-12-17:
can you contact me over facebook messenger?

can you just type

python3

import tkinter
tkinter.Tk()
exit()

and see what it does

wnc2, 2020-12-17:
I'M HALF THE WAY!!:

unistalled all, 3.8, 3.9, homebrew etc
reinstalled latest 3.9.1
i ran:
python3

import tkinter
tkinter.Tk()
exit()

ok a little empty window opened in the upper left screen.

i ran the p2pp.command and all ok
[image]

but then if i run from PS i still have errors
[image]

could you confirm latest p2pp.command file?

in the meanwhile i noticed another 5.27 version i try to find it

tomvandeneede, 2020-12-17:
this is good!!!

wnc2, 2020-12-17:
half good :-)

> this is good!!!

audiorazor, 2020-12-17:
woo, I got the little box to pop up via the command line also

audiorazor, 2020-12-17:
Do I want to install 5.27 or is that not ready yet?

tomvandeneede, 2020-12-17:
problem remains…. how to explain to the masses… this is not something very intuitive

wnc2, 2020-12-17:
> Do I want to install 5.27 or is that not ready yet?

i don't find it for mac

tomvandeneede, 2020-12-17:
you can… just fixed an issue today… guess there will be a few more, 4.16 is the old stable version, lot slower and memory hungry though

tomvandeneede, 2020-12-17:
sorry, just patched the bug but did not yet update the pac installer

tomvandeneede, 2020-12-17:
the new version will have a pruse2.3 and later configurator
it should take some of the work out of the equation
should be ready for testing in a few days

it will just do the first setup with the basics… the special processing will still be manual

wnc2, 2020-12-17:
> problem remains…. how to explain to the masses… this is not something very intuitive

it seems:
1) install plain python 3.9.1 package (ora latest)
2) install P2PP

[25 more replies]

closed 2020-12-20 by tomvandeneede