tomvandeneede/p2pp #83: on DMG version g-code doesn't intercepts palette

wnc2, 2020-12-19:
if i run a script, all goes ok, but when i run the gcode from octoprint, the printers start to print without running palette, octoprint shows "waiting for palette... ecc ecc" but nothing happens, palette remains in standby, and the printer start to print by itself

tomvandeneede, 2020-12-19:
There seems to be an issue with send to printer and PrusaSlicer 2.3.   In 2.2 all works just fine, in 2.3 I get brief pop-up of P2PP and then it dies without processing.  in simple export mode it works just fine.   Looking if this could be something in Prusaslicer

wnc2, 2020-12-19:
the issue seems random

tomvandeneede, 2020-12-19:
well I added a check to latest version of P2PP app that stops processing if the file is not found
It appears 1/2 of the time the parameter passed to the post processing script is not opening properly/

Still need to figure out if it has anything to do with p2pp or if it is something in PS2.3

wnc2, 2020-12-19:
6.00.002?

tomvandeneede, 2020-12-19:
Yes

below you see the parameter I am getting from PrusaSlicer and the files that I find in that folder.
The file is not there so there must be something wrong on their side.    If I don’t get the right name it is difficult to process

wnc2, 2020-12-19:
ok, i already installed 002 before, in fact last print is gone to palette

tomvandeneede, 2020-12-19:
they all go to printer but if they don’t get processed they don’t print as they are supposed to

wnc2, 2020-12-19:
yesterday i told you i have strange delays sending from PS to octoprint (prin host upload queue), you told me the script is not involved, are you sure? never had before

tomvandeneede, 2020-12-19:
well the printer can only start uploading after the P2PP window closes (this is when the control returns to PS2.3
I don’t see why we would still influence uploading after P2PP closed

it could be when they start uploading BEFORE P2PP ends but not sure what code they would be using then

tomvandeneede, 2020-12-20:
relates to unasync calling of the script leading to the file being stolen away before it could be opened

closed 2020-12-20 by tomvandeneede