tomvandeneede/p2pp #84: MultiExtruder gcode is generated even if only one is configured (M572 codes are wrong)

BEE-KA, 2020-12-27:
P2PP Version: 6.01.07
Python Version: 3.9.1
PrusaSlicer Version 2.3.0-rc2+win64
Windows 10

3D printer brand / version + firmware version (if known)
BLV MGN Duett Wifi with RepRap

Behavior of PrusaSlicer:
Gcode for multiple Extruder is generated instead of only one.
I use the following configuration for my printer with Mosaic Palette 2 pro:
image
The result in Gcode is a M572 S0 D0 code an also
M572 S0 D1 (for second extruder) but i do not configured a second.
Can you please check this? Is it is possible to delete or change this lines by p2pp?
I expect always a M572 S0 D0 commant (extruder 1) in generated gcode.

tomvandeneede, 2020-12-27:
This is a configuration error on the prusa side.  You need to set the single extruder multilateral glad iunder printer settings/Generaland this behaviour should go away
I prefer not to handle this in P2PP as it required further in depth analysis of the code which might need a lot more changes than just the tool change you see

BEE-KA, 2020-12-27:
[image]
I did this...

tomvandeneede, 2020-12-27:
should be very much ok then
can you share the gcode file

tomvandeneede, 2020-12-28:
Added M572 processing to P2PP as of version 6.2.1

BEE-KA, 2021-01-05:
Thank you verry much

closed 2022-02-05 by tomvandeneede